schollz/find #5: Always classified as currently "unknown"

kootenpv, 2015-11-22:
Using "bed" and "couch" as positions, after recalculating priors, still always unknown position. This happens both using the online demo as well as running my own server. Here a dump, in case that might help:

11-22 10:58:08 routing-fingerprinting DEBUG    got POST from pas (pasje) posting bed fingerprint to /track
MODIFIED ROUTINE TOOK 0.0014650821685791016
11-22 10:58:08 posteriors.processTrackingFingerprint WARNING  No GPX for bed
11-22 10:58:08 posteriors.processTrackingFingerprint DEBUG    [('bed', nan), ('big couch', nan)]
11-22 10:58:08 posteriors.processTrackingFingerprint WARNING  pas (pasje) at bed with prob ratio: nan/nan
{'message': 'Saved to database /track', 'position': [('bed', nan), ('big couch', nan)], 'success': True}
11-22 10:58:08 tornado.access INFO     200 POST /track (192.168.2.2) 4.84ms
11-22 10:58:11 routing-fingerprinting DEBUG    got POST from pas (pasje) posting bed fingerprint to /track
11-22 10:58:11 posteriors.processTrackingFingerprint DEBUG    adding 98:f5:37:52:5a:76
11-22 10:58:11 posteriors.processTrackingFingerprint DEBUG    adding 62:02:92:49:1b:f8
11-22 10:58:11 posteriors.processTrackingFingerprint DEBUG    adding 00:14:7f:76:51:39
MODIFIED ROUTINE TOOK 0.0011620521545410156
11-22 10:58:11 posteriors.processTrackingFingerprint WARNING  No GPX for bed
11-22 10:58:11 posteriors.processTrackingFingerprint DEBUG    Saving bulk tracking data for pas
11-22 10:58:11 mldatabase:insertFingerprint DEBUG    Successfully inserted 2 fingerprints to "track"
11-22 10:58:11 posteriors.processTrackingFingerprint DEBUG    [('bed', nan), ('big couch', nan)]
11-22 10:58:11 posteriors.processTrackingFingerprint WARNING  pas (pasje) at bed with prob ratio: nan/nan
{'message': 'Saved to database /track', 'position': [('bed', nan), ('big couch', nan)], 'success': True}
11-22 10:58:11 tornado.access INFO     200 POST /track (192.168.2.2) 6.76ms
11-22 10:58:13 routing-fingerprinting DEBUG    got POST from pas (pasje) posting bed fingerprint to /track
MODIFIED ROUTINE TOOK 0.0014197826385498047
11-22 10:58:13 posteriors.processTrackingFingerprint WARNING  No GPX for bed
11-22 10:58:13 posteriors.processTrackingFingerprint DEBUG    [('bed', nan), ('big couch', nan)]
11-22 10:58:13 posteriors.processTrackingFingerprint WARNING  pas (pasje) at bed with prob ratio: nan/nan
{'message': 'Saved to database /track', 'position': [('bed', nan), ('big couch', nan)], 'success': True}
11-22 10:58:13 tornado.access INFO     200 POST /track (192.168.2.2) 4.90ms
11-22 10:58:15 routing-fingerprinting DEBUG    got POST from pas (pasje) posting bed fingerprint to /track
11-22 10:58:15 posteriors.processTrackingFingerprint DEBUG    adding 6a:f5:37:52:5a:77
11-22 10:58:15 posteriors.processTrackingFingerprint DEBUG    adding ee:55:f9:7b:0f:dd
11-22 10:58:15 posteriors.processTrackingFingerprint DEBUG    adding 9c:80:df:d0:8c:a0
11-22 10:58:15 posteriors.processTrackingFingerprint DEBUG    adding 52:95:7f:50:cc:e7
11-22 10:58:15 posteriors.processTrackingFingerprint DEBUG    adding 60:02:92:49:1b:f7
MODIFIED ROUTINE TOOK 0.0011739730834960938
11-22 10:58:15 posteriors.processTrackingFingerprint WARNING  No GPX for bed
11-22 10:58:15 posteriors.processTrackingFingerprint DEBUG    Saving bulk tracking data for pas
11-22 10:58:15 mldatabase:insertFingerprint DEBUG    Successfully inserted 2 fingerprints to "track"
11-22 10:58:15 posteriors.processTrackingFingerprint DEBUG    [('bed', nan), ('big couch', nan)]
11-22 10:58:15 posteriors.processTrackingFingerprint WARNING  pas (pasje) at bed with prob ratio: nan/nan
{'message': 'Saved to database /track', 'position': [('bed', nan), ('big couch', nan)], 'success': True}
11-22 10:58:15 tornado.access INFO     200 POST /track (192.168.2.2) 7.09ms
11-22 10:58:17 routing-fingerprinting DEBUG    got POST from pas (pasje) posting bed fingerprint to /track
11-22 10:58:17 posteriors.processTrackingFingerprint DEBUG    adding 62:02:92:49:1b:f8
MODIFIED ROUTINE TOOK 0.0014040470123291016
11-22 10:58:17 posteriors.processTrackingFingerprint WARNING  No GPX for bed
11-22 10:58:17 posteriors.processTrackingFingerprint DEBUG    [('bed', nan), ('big couch', nan)]
11-22 10:58:17 posteriors.processTrackingFingerprint WARNING  pas (pasje) at bed with prob ratio: nan/nan
{'message': 'Saved to database /track', 'position': [('bed', nan), ('big couch', nan)], 'success': True}
11-22 10:58:17 tornado.access INFO     200 POST /track (192.168.2.2) 5.10ms
11-22 10:58:19 routing-fingerprinting DEBUG    got POST from pas (pasje) posting bed fingerprint to /track
11-22 10:58:19 posteriors.processTrackingFingerprint DEBUG    adding 84:18:3a:3b:e9:b8
MODIFIED ROUTINE TOOK 0.0013890266418457031
11-22 10:58:19 posteriors.processTrackingFingerprint WARNING  No GPX for bed
11-22 10:58:19 posteriors.processTrackingFingerprint DEBUG    Saving bulk tracking data for pas
11-22 10:58:19 mldatabase:insertFingerprint DEBUG    Successfully inserted 2 fingerprints to "track"
11-22 10:58:19 posteriors.processTrackingFingerprint DEBUG    [('bed', nan), ('big couch', nan)]
11-22 10:58:19 posteriors.processTrackingFingerprint WARNING  pas (pasje) at bed with prob ratio: nan/nan
{'message': 'Saved to database /track', 'position': [('bed', nan), ('big couch', nan)], 'success': True}
11-22 10:58:19 tornado.access INFO     200 POST /track (192.168.2.2) 6.71ms

kootenpv, 2015-11-22:
It is now working, the current position is correct! Just wondering...it says "real time" current position, but it takes a long time for it to display that it switched a new location (something like a minute). 

What is the update interval?

schollz, 2015-11-22:
Hmmm. It could be that server was busy - if someone is doing a "Calculate All and Find Parameters" you would have to wait until it finished. Have you been experiencing this frequently? 

The polling depends on the device. On the app you can set it up to poll every 2 seconds, so the "real time" position should update every 2-4 seconds.

Also, are you still getting 'nan' for your positions? You should be getting numbers after you've done some learning.

schollz, 2016-04-16:
Returns null

closed 2016-04-16 by schollz