schollz/find #7: Closed source component disclosure/inform

assimilat, 2015-12-28:
Hi!
Just wanna start by saying I love the project! 
    I just did a quick look over on the code/install process and I noticed the setup script downloads a package of go compiled binaries (calculate.zip) that im assuming (thanks to the name and size) is the brains of the operation. Looking at internalpositioning.com it says:

 *"We have kept the useful part of FIND open-source, so that you can use it for your own projects. "* 

...followed by a link to this repo.

Dont get me wrong all of that is great, but you might want to not say that the code hosted here is open source without at least mentioning that the setup process downloads a (*seemingly pretty important*) closed source component or maybe add a prompt in the installer to alert the user that the download is about to happen...or better yet open the source ;-)?
Thanks for all the hard work!
A

schollz, 2016-01-02:
Thanks! 

You're right, there is one bit of the code that is not "open"... this is the code that has the math behind the classification algorithm. 

The classification algorithm is compiled in an attempt to obfuscate how it works. Even though the algorithm is solely and unequivocally my own original creation, there are currently tons of patents being submitted about Wifi-based positioning. I don't want to be responsible for possibly causing people to accidentally infringe on patent-holders, so I opted to make it more difficult for a patent holder to find my exact implementation. I suppose I could check all the patents to make sure I'm not infringing, but I rather stay ignorant to what they've done in order to better my case that mine is an original work.

As for your suggestion for the prompt, its a good one, and I'll add it. I'll also definitely add the disclaimer.

assimilat, 2016-01-15:
Awesome! Thanks for the reply!

closed 2016-01-15 by assimilat