schollz/find #8: Roadmap

kootenpv, 2016-01-06:
Where will this go?

schollz, 2016-01-07:
There are tons of short term things I'm working on (adding delete for users, allowing custom names for groups, better D3 charts, automation to machine learning, removing platform specific commands).

As for long term, I'm hoping to get more GIS elements integrated and have a more User-friendly system. The current repo has a tiny bit of GIS stuff (see the routes map2.html and mappingdata.html, which uses a 3rd party mechanism). Currently I'm working with a friend on these aspects, with a much better large-scale implementation (using Django for the system admin and Postgres and Leaflet for the mapping stuff). 

My main long term goal, and my motives for posting this on Github, was to try to integrate **find** for applications that other folks were interested. I'm especially interested in home automation, and I've tried to tie that in to the routing system. There are plenty of other things that **find** could be used for - peer-to-peer wayfinding, path prediction...there is lots of room for improvement in this area and lots of potential, I hope.

I'll think about a more exhaustive roadmap and post it to the main readme soon!

kootenpv, 2016-01-07:
Great, thanks a lot!

closed 2016-01-07 by kootenpv