schollz/find #14: Add in OS X for fingerprint CLI program

schollz, 2016-04-16:
This is an easy task, but I can't do it because I don't have a Mac.

Basically you'll need to run

```
/System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport -I en0
```

And parse the output to collect only the BSSIDs and the Signal strengths.

First, I'd be really happy if **you can run that command and post the output here!**

Second, I'd be thrilled if you'd like to contribute and [write the parser](https://github.com/schollz/find/blob/fingerprint/main.go#L65-L68)

schollz, 2016-04-17:
```
$ sudo /System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport --scan
                            SSID BSSID             RSSI CHANNEL HT CC SECURITY (auth/unicast/group)
                        Colombia 2c:b0:5d:33:52:02 -71  6       Y  -- WPA2(PSK/AES/AES) 
                      ARRIS-92AA 5c:8f:e0:52:92:a8 -86  6       Y  US WPA2(PSK/AES/AES) 
                         scp_pri 2e:a4:3c:3f:c7:36 -79  6       Y  -- WPA(PSK/AES/AES) WPA2(PSK/AES/AES) 
                        Blessing 5c:8f:e0:52:22:a8 -72  11      Y  US WPA2(PSK/AES/AES) 
               WATER STREET 2918 48:f8:b3:2a:36:00 -73  11      Y  -- WPA2(PSK/AES/AES) 
                            2917 5c:8f:e0:51:35:a0 -58  11,-1   Y  US WPA2(PSK/AES/AES) 
                        Cisco 2G c8:b3:73:49:e1:bd -68  9       Y  -- WPA(PSK/AES,TKIP/TKIP) WPA2(PSK/AES,TKIP/TKIP) 
                   ARRIS-2803 5G 10:86:8c:0f:42:cd -87  157,+1  Y  US WPA2(PSK/AES/AES) 
                     Linksys2718 b4:75:0e:ab:a4:61 -73  6       Y  -- WPA(PSK/AES,TKIP/TKIP) WPA2(PSK/AES,TKIP/TKIP) 
                          Lipsig 04:a1:51:d7:a3:fb -74  4       Y  -- WPA2(PSK/AES/AES) 
                      The 69ers_ b4:75:0e:58:db:9b -77  3       Y  -- WPA2(PSK/AES/AES) 
                         Drewslo 48:f8:b3:3a:47:26 -75  1       Y  -- WPA(PSK/AES,TKIP/TKIP) WPA2(PSK/AES,TKIP/TKIP) 
                          George 58:6d:8f:c4:8e:e6 -81  1       Y  -- WPA(PSK/AES,TKIP/TKIP) WPA2(PSK/AES,TKIP/TKIP) 
                  Patriot League b4:75:0e:ac:29:45 -75  1       Y  -- WPA2(PSK/AES/AES) 
                           Jcraw 48:f8:b3:3a:62:06 -45  1       Y  -- WPA2(PSK/AES/AES) 
                     Colombia-5G 2c:b0:5d:33:52:01 -83  153,-1  Y  -- WPA2(PSK/AES/AES) 
                  The 69ers PLUS b4:75:0e:58:db:9c -85  149,+1  Y  -- WPA2(PSK/AES/AES) 
                        Cisco 5G c8:b3:73:49:e1:be -79  149,+1  Y  -- WPA(PSK/AES,TKIP/TKIP) WPA2(PSK/AES,TKIP/TKIP) 
                         KSBR-5G 14:cf:e2:8d:32:15 -85  44,+1   Y  US WPA2(PSK/AES/AES) 
                    DG1670A72-5G 90:1a:ca:83:28:75 -75  44,+1   Y  US WPA2(PSK/AES/AES) 
               Patriot League 5G b4:75:0e:ac:29:46 -82  40,-1   Y  -- WPA2(PSK/AES/AES) 
                           2917* 5c:8f:e0:51:35:a5 -78  36,+1   Y  US WPA2(PSK/AES/AES) 
                     Blessing-5G 5c:8f:e0:52:22:ad -85  161,-1  Y  US WPA2(PSK/AES/AES) 
          PoliceSurveillanceVan1 00:19:e3:fa:46:af -44  157,+1  Y  US WPA2(PSK/AES/AES) 
```

jimmed, 2016-04-18:
Here's the output from the original command provided:

```
jimobrien ~> /System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport -I en0
     agrCtlRSSI: -76
     agrExtRSSI: 0
    agrCtlNoise: -92
    agrExtNoise: 0
          state: running
        op mode: station
     lastTxRate: 24
        maxRate: 54
lastAssocStatus: 0
    802.11 auth: open
      link auth: wpa2-psk
          BSSID: c0:3e:f:2f:c3:ad
           SSID: Dig
            MCS: -1
        channel: 4
```

This didn't seem to provide the information you were after, so I checked the help, and instead used `-s`/`--scan` to produce this:

```
jimobrien ~> /System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport -s
                            SSID BSSID             RSSI CHANNEL HT CC SECURITY (auth/unicast/group)
                BTHub5-KZPF 5GHZ 00:91:f9:61:01:af -80  44      Y  GB WPA2(PSK/AES/AES)
                 BTWifi-with-FON c2:91:f9:61:01:ae -73  1       Y  GB NONE
                          MopNo2 c0:3f:0e:70:ec:f2 -66  2       N  -- WPA(PSK/TKIP,AES/TKIP) WPA2(PSK/TKIP,AES/TKIP)
                     BTHub5-7R6P 00:8a:ae:b7:30:d0 -87  1       Y  GB WPA2(PSK/AES/AES)
                       SKYLINKW1 60:a4:4c:68:d6:90 -88  1       Y  -- WPA2(PSK/AES/AES)
                     BTHub5-KZPF 00:91:f9:61:01:ae -69  1       Y  GB WPA2(PSK/AES/AES)
                        SKY4370B 90:21:06:14:1f:e1 -62  1       Y  DE WPA2(PSK/AES/AES)
                        BTWifi-X 32:1e:78:2e:fc:ba -73  11      Y  GB WPA(802.1x/AES,TKIP/TKIP) WPA2(802.1x/AES,TKIP/TKIP)
                 BTWifi-with-FON 12:1e:78:2e:fc:ba -71  11      Y  GB NONE
                     BTHub5-TCNC 00:1e:78:2e:fc:ba -73  11      Y  GB WPA2(PSK/AES/AES)
                        SKYDB562 c0:3e:0f:97:d9:45 -81  6       Y  DE WPA2(PSK/AES/AES)
                       SKYLINKW2 f4:06:8d:06:32:ba -79  6       Y  -- WPA2(PSK/AES/AES)
                             Dig c0:3e:0f:2f:c3:ad -69  4       N  DE WPA2(PSK/AES/AES)
```

This is on OSX El Capitan 10.11.4 (15E65) on an early 2015 MBP.

schollz, 2016-04-18:
Great, thanks. 
So the command I'll use is

```
/System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport -s
```

I'll put together a parser for that output.

bebus77, 2016-04-18:
I can write the parser if you haven't already.

schollz, 2016-04-18:
I haven't, so go for it! There are tests in there, just follow my lead in `windows_test.go` and `linux_test.go`.

Pugio, 2016-04-18:
Wanted to try it out myself so wrote #37 .

schollz, 2016-04-18:
Thanks https://github.com/Pugio. Its now merged.

closed 2016-04-18 by schollz