schollz/find #20: Redo /whereami route for GIS server

schollz, 2016-04-16:


sjsafranek, 2016-04-16:
{
 username: [location_name]
}

sjsafranek, 2016-04-16:
{
   username: [
      Location_name: {
         <machine learning properties>
      }
   ]
}

First item in the array of locations in the last location for that user

schollz, 2016-04-16:
`/location?group=GROUP&history=X&user=USER`

`history` and `user` are optional. `history` defaults to 1. `user` defaults to all users.

```json
{
    "user": [
        {
            "time": "2016-04-16 08:16:43.123233725 -0400 EDT",
            "location": "",
            "bayes": {}
        }
    ]
}
```

schollz, 2016-04-21:
https://github.com/schollz/find/commit/89d389a93ca8fb78c22ea761a692f5400f3c5f5b

/location?group=GROUP&user=user1
/location?group=GROUP&users=user1,user2

closed 2016-04-21 by schollz