schollz/find #32: Go get to install find

lafolle, 2016-04-18:
It would be great if we could just `go get github.com/schollz/find` to install `find` rather than 3 steps to manually build.

2 problems will arise because of this:
1. name collision with existing find (linux) command
2. relative paths are used to specify static content for dashboard. (easier to fix)

Thanks,
lafolle

schollz, 2016-04-18:
Lol I forgot about the name collision. I could change the name. Since its written in Golang it could be "gofind". 

For (2), there are a couple of solutions. The easiest is to fix the path issue so that it still points to the static content. In reality though, I use NGINX/Caddy to serve the static content so it doesn't matter where the program thinks it is as it never pulls it (though you would need to define the path to DB still). I think I'd rather try to build the static content into the binary using gorice or something. I've never done that and would like to learn it.

lafolle, 2016-04-19:
You can also check this out => https://github.com/elazarl/go-bindata-assetfs

PS: In case you've don't find time, i'll be glad to help. :)