schollz/find #33: License

fabiencastan, 2016-04-18:
Hi,

I just noticed that you changed the license from apache to gpl or agpl: https://github.com/schollz/find/commit/45186a15c66c01870c5ae119a212dad34e96e9a1
In the [README](https://github.com/schollz/find/blob/master/README.md) it's written GPL but in the [license file](https://github.com/schollz/find/blob/master/LICENSE) it's written AGPL.

Is there a chance to keep this great library under a license without a strong copyleft, like [LGPL](https://en.wikipedia.org/wiki/GNU_Lesser_General_Public_License) or [MPL](https://en.wikipedia.org/wiki/Mozilla_Public_License)?

Best Regards,
Fabien

schollz, 2016-04-18:
Hi Fabien,

Thanks for pointing out the inconsistency. Right now everything is meant to be on AGPL. What is the benefit of moving to something like LGPL or MPL?

TheToolbox, 2016-04-18:
Some licenses (GPL-based especially) have strong requirements that any users of the covered material must release the source code (under the same license) of any software that they build with the GPL-covered code. LGPL licenses allow closed-source and profit-generating software to be built on top of the LGPL-covered code.

fabiencastan, 2016-04-18:
Hi,

1/ Using AGPL means that all code that interact directly or indirectly with FIND should be release under the AGPL license.
2/ Using LGPL or MPL means that all modifications of FIND itself should be released under the same license but enables to use FIND in a software with another license.

The first one ensures that only open-source softwares using the same license could use it. The goal of these licenses is to ensure the freedoms are preserved whenever the work is used.
The second solution ensures that the usage of the library is fair: all modifications should be given back to the community but offers the freedom to use it in all contexts. The goal of these licenses is to maximize the number of users and contributors.

In my company, most of our developments are in open-source but we also use and develop proprietary softwares and we also develop extensions to proprietary softwares. If FIND use a strong copyleft license, we will not be able to use it.

Thanks,

pizzamaker, 2016-04-18:
Personally, I support AGPL as it closes the infamous web app loophole. It's generally recommended for any software that will commonly be run over a network, such as FIND.

schollz, 2016-04-19:
Thanks for the clarification. That's really helpful. I'm going to keep it AGPL for now, I've fixed the inconsistencies in the README to reflect this.

> The goal of these licenses is to maximize the number of users and contributors.

I agree, and that is what I hope to do with FIND. However, before I change over to these licenses I will need to consult. 

@fabiencastan, please feel free to contact me at zack@hypercubeplatforms.com and we could work out something so your company could use it if you're still interested!

fabiencastan, 2016-04-19:
Sure, it really depends on the objectives and contexts of development.
Thanks for the feedback!

closed 2016-04-19 by fabiencastan