schollz/find #36: index out of range in /calculate

kolen, 2016-04-18:
For me, occurs always, each time I press "Database statistics → Update".

```
2016/04/18 18:18:39 [Recovery] panic recovered:
GET /calculate?group=taburet HTTP/1.1
Host: 192.168.1.2:8003
Accept: */*
Accept-Encoding: gzip, deflate, sdch
Accept-Language: ru-RU,ru;q=0.8,en-US;q=0.6,en;q=0.4
Connection: keep-alive
Cookie: mysession=MTQ2MDk5MjU5OXxEdi1CQkFFQ180SUFBUkFCRUFBQUp2LUNBQUVHYzNSeWFXNW5EQWNBQldkeWIzVndCbk4wY21sdVp3d0pBQWQwWVdKMWNtVjB8P4MMxlsJmvDPI4v4zL1oG6EpDvYpdz-7rg6lX0VF0OE=
Referer: http://192.168.1.2:8003/dashboard/taburet
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_11_4) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/50.0.2661.66 Safari/537.36
X-Requested-With: XMLHttpRequest

runtime error: index out of range
/usr/local/Cellar/go/1.6.1/libexec/src/runtime/panic.go:426 (0x748e9)
	gopanic: reflectcall(nil, unsafe.Pointer(d.fn), deferArgs(d), uint32(d.siz), uint32(d.siz))
/usr/local/Cellar/go/1.6.1/libexec/src/runtime/panic.go:15 (0x72db9)
	panicindex: panic(indexError)
/Users/kolen/items/gopath/find/priors.go:229 (0x35678)
	calculatePriors: ps.Priors[networkName].P[v2.Location][router.Mac][router.Rssi-MinRssi] += PdfType[0]
/Users/kolen/items/gopath/find/priorsThreaded.go:83 (0x3633a)
	optimizePriorsThreaded: calculatePriors(group, &ps, fingerprintsInMemory, fingerprintsOrdering)
/Users/kolen/items/gopath/find/api.go:79 (0x32d7)
	calculate: optimizePriorsThreaded(strings.ToLower(group))
/Users/kolen/items/gopath/src/github.com/gin-gonic/gin/context.go:97 (0x17dfea)
	(*Context).Next: c.handlers[c.index](c)
/Users/kolen/items/gopath/src/github.com/gin-gonic/contrib/sessions/sessions.go:65 (0x1e8733)
	Sessions.func1: c.Next()
/Users/kolen/items/gopath/src/github.com/gin-gonic/gin/context.go:97 (0x17dfea)
	(*Context).Next: c.handlers[c.index](c)
/Users/kolen/items/gopath/src/github.com/gin-gonic/gin/recovery.go:45 (0x190881)
	RecoveryWithWriter.func1: c.Next()
/Users/kolen/items/gopath/src/github.com/gin-gonic/gin/context.go:97 (0x17dfea)
	(*Context).Next: c.handlers[c.index](c)
/Users/kolen/items/gopath/src/github.com/gin-gonic/gin/logger.go:63 (0x18fb1a)
	LoggerWithWriter.func1: c.Next()
/Users/kolen/items/gopath/src/github.com/gin-gonic/gin/context.go:97 (0x17dfea)
	(*Context).Next: c.handlers[c.index](c)
/Users/kolen/items/gopath/src/github.com/gin-gonic/gin/gin.go:284 (0x184e42)
	(*Engine).handleHTTPRequest: context.Next()
/Users/kolen/items/gopath/src/github.com/gin-gonic/gin/gin.go:265 (0x184a77)
	(*Engine).ServeHTTP: engine.handleHTTPRequest(c)
/usr/local/Cellar/go/1.6.1/libexec/src/net/http/server.go:2081 (0x10b9ee)
	serverHandler.ServeHTTP: handler.ServeHTTP(rw, req)
/usr/local/Cellar/go/1.6.1/libexec/src/net/http/server.go:1472 (0x10829e)
	(*conn).serve: serverHandler{c.server}.ServeHTTP(w, w.req)
/usr/local/Cellar/go/1.6.1/libexec/src/runtime/asm_amd64.s:1998 (0xa4491)
	goexit: BYTE	$0x90	// NOP
```

schollz, 2016-04-18:
Thanks! Two questions:

How many different locations did you fingerprint?

How many fingerprints (approx) are learned for each different location?

If you want, you could email me your database (data/taburet.db) to (zack@hypercubeplatforms.com) so I could easily reproduce the error.

schollz, 2016-04-18:
Okay, this problem is because of the RSSI value was -9 and min MinRssi is -10. I will submit a commit to lower the MinRssi to accomodate this.

pablomusumeci, 2016-04-21:
Same for me. I've fingerprinted 3 locations with 50 fingerprints for each one.

schollz, 2016-04-22:
I've increased the MaxRssi, so you should not have this issue anymore. Let me know!

pablomusumeci, 2016-04-22:
This index out of range error keeps happening but in a different line of code (priors.go:233) :

`calculatePriors: ps.Priors[networkName].P[v2.Location][router.Mac][router.Rssi-MinRssi+i] += val`

schollz, 2016-04-22:
Okay, so I think you are somehow getting RSSI values < -100, which I thought shouldn't be possible.

To be sure, could you please send me your group name if your using the server, or email your data/GROUPNAME.db file? My email is zack@hypercubeplatforms.com.

schollz, 2016-04-22:
Thanks! Looks like I was wrong. You had a value of -4 that was slightly out of range as it was too strong still. So I made the buffer a little bigger (now will accept impossible values of at most 1). 

Please check to see that it works now.

pablomusumeci, 2016-04-22:
Thanks Zach. I will test it again tonight and let you know if it worked.

commit bb59588d (2016-04-22): Increase maxRSSI a little more to fix #36. Now should be good