schollz/find #42: Scan failed, will continue after a rest

jimdsouza, 2016-04-18:
I get the following error while trying to learn on my Raspberry Pi 2B.

`00:43:40.694 main - NOTI 006 Running 3 times (you can run more using '-c SOMENUM'). Please wait...
00:43:40.694 main - INFO 007 Scanning Wifi
00:43:40.694 scanWifi - INFO 008 Gathering fingerprint with '/sbin/iw dev wlan0 scan -u'
00:43:40.702 main - WARN 009 Scan failed, will continue after a rest
00:43:43.703 main - INFO 00a Scanning Wifi
00:43:43.704 scanWifi - INFO 00b Gathering fingerprint with '/sbin/iw dev wlan0 scan -u'
00:43:43.715 main - WARN 00c Scan failed, will continue after a rest
00:43:46.717 main - INFO 00d Scanning Wifi
00:43:46.718 scanWifi - INFO 00e Gathering fingerprint with '/sbin/iw dev wlan0 scan -u'
00:43:46.728 main - WARN 00f Scan failed, will continue after a rest
`

I tried `sudo /sbin/iw dev wlan0 scan -u` and it says **nl80211 not found.**

schollz, 2016-04-18:
You might try `sudo apt-get install bridge-utils` and then run `/sbin/iw dev wlan0 scan -u` again. 

What kind of Wifi adapter are you using?

jimdsouza, 2016-04-18:
I just installed that and am still getting the same error. I'm using one of those [TP-Link USB network adapters](http://www.tp-link.com/en/products/details/TL-WN823N.html) on my PI. The driver is **rtl8192cu**.

schollz, 2016-04-18:
What OS? Raspbian?

jimdsouza, 2016-04-18:
Yes.

schollz, 2016-04-18:
It may be that your specific TP-link adapter doesn't have the right firmware to support `iw list`. 

Here's something I found that has update to firmware: https://www.linuxquestions.org/questions/linux-hardware-18/wifi-help-debian-7-and-usb-tp-link-tl-wn725n-4175460688/

This one seems to suggest you can edit your `hostap` file to get it working: http://jacobsalmela.com/raspberry-pi-and-routing-turning-a-pi-into-a-router/

To be clear, I've only tested this on Raspberry Pi B+/2 w/ the Edimax donle and with Raspberry Pi 3 built-in on Raspbian. Both worked for me right out of the box. So I'm not sure I can help you with yours. Please let me know if you find a solution!

jimdsouza, 2016-04-21:
The links you provide are for installing the driver on Pi. However, my Pi is connected to the internet via the same adapter and `ifconfig` does show wlan0.

Only similarity is that `iw wlan0 info` returns nl80211 not found. So I'm guessing, Pi has a driver since it's using it to connect to the network , but **iw** isn't recognising it.

TL-WN823N also worked on my Pi 2B out of the box.

schollz, 2016-04-21:
Could you see if this command works: `sudo iwlist wlan0 scan` ?

jimdsouza, 2016-04-21:
Yes, it works.

schollz, 2016-04-21:
Fantastic! I'll work on an update to the fingerprint client that supports iwlist.

jimdsouza, 2016-04-21:
Thank you. The [earlier link](https://www.raspberrypi.org/forums/viewtopic.php?&t=18745) says iwconfig can be used as an alternative to iw too.

schollz, 2016-04-21:
Here's the new issue: https://github.com/schollz/find/issues/62

I'll close this when that gets built.

schollz, 2016-04-21:
Okay its fixed now - it will work with `iwlist` and `iw`. You can switch using a CLI flag `--iwlist`, and the default is `iw` still. You can build from source or download a pre-release here: https://github.com/schollz/find/releases/tag/v0.4client.

schollz, 2016-04-21:
Actually, please confirm this works for you before I close! thanks :)

jimdsouza, 2016-04-21:
Hi. Thanks for the quick fix. But I used the prebuilt v0.4 on my Pi and `sudo ./fingerprint -e -g xxxx -l yyyy` gives the same error. The only change is, it returns "false" and then proceeds to use iw like before.

jimdsouza, 2016-04-21:
Thanks for that quick fix. It works perfectly now with the `--iwlist` flag.

closed 2016-04-21 by schollz