schollz/find #49: Android: Intent or Tasker integration to start/stop tracking

Christoph-Wagner, 2016-04-19:
This would be my feature request :)

My scenario is, that I'd like to start tracking for a few seconds when connected to my wifi and the accelerometer activates.
I read the information from the find server into my home-automation system and get room-level tracking (tested it, it works well enough for my purposes) to turn on the lights and so on.

A nice to have would be a configurable tracking interval, but I guess I could do that myself even without android experience and recompile it ;)

schollz, 2016-04-19:
> A nice to have would be a configurable tracking interval
https://github.com/schollz/find/issues/50

schollz, 2016-04-19:
Tasker integration would be awesome! 

Here's the [Tasker doc for getting this started](http://tasker.dinglisch.net/developers.html). It doesn't look so bad. If someone has experience with this, please let me know!

schollz, 2016-04-19:
I just realized a primitive way of doing this, that might work, would just be to create a new FIND app that starts tracking right when it opens. That way you could use Tasker -> Start App/Kill App to get it to just open the app and the app will start tracking right away. I will play with this, and if it works, I'll release this version as a new app on the play store so you could play with it.

Christoph-Wagner, 2016-04-19:
Just for playing with it, I can already use another Tasker plugin that just clicks on the Start Tracking button. So don't feel the need to release that on my account ;) Of course if no one gets around to tasker integration it might still make sense, but tasker commands are cleaner :)

Christoph-Wagner, 2016-04-20:
The [intent plugin](https://www.npmjs.com/package/cordova-plugin-intent) seems like an extremely simple alternative to set up a track or learn intent. Full tasker integration would be better but this should be far easier to implement.