schollz/find #50: Make tracking interval adjustable

schollz, 2016-04-19:
Need to add Button for the Options screen in https://github.com/schollz/find/blob/android/index.html

Add new variable for tracking interval and prompt in https://github.com/schollz/find/blob/android/main.js

Christoph-Wagner, 2016-04-20:
I'll start on this.

schollz, 2016-04-20:
Thanks. 

Note, that if you are building there is a gotcha in the Cordova system. Cordova 5.X should be fine, but the newest Cordova 6.X defaults to android-23. Unfortunately, android-23 requires Runtime permissions for the WiFi access stuff and currently I haven't figured out how to do Runtime permissions. So if the build will succeed for android-23 but it won't work. To get around this you can change project.properties and the manfiest to android-22. However, in the future I'll be trying to add in those Runtime permissions....

Christoph-Wagner, 2016-04-20:
Would it be okay to send the PR just having tested that the change works in platform browser? I'd like to avoid installing the whole android studio :/

schollz, 2016-04-20:
Definitely. Its easy for me to test, so its no problem.

schollz, 2016-04-20:
Done in https://github.com/schollz/find/commit/5ccc0997da1861ffe7986dc9499ad41b2d59a61d
Thanks Christoph-Wagner!

closed 2016-04-20 by schollz