schollz/find #52: Offline data collection sometimes loses data in Android app

kolen, 2016-04-19:
Sometimes Android app loses data collected during time where there was no connection to server. Sometimes it works correctly and uploads data later.

schollz, 2016-04-19:
Sorry, I think this may have been due to a couple of server restarts where the server was down. It shouldn't be happening anymore.

To answer your comment - the app loses the data if there is no connection. It doesn't wait until the server is backup. I'm not necessarily sure I want to design something that assumes the server will be offline every once in awhile, because nominally it never will. But please let me know if you are continually experiencing this and I can implement a fix.

kolen, 2016-04-19:
I used my own server that was not restarted.

> To answer your comment - the app loses the data if there is no connection.

But once I was able to record data for several places without connection. My server was in internal network and not available from internet but I recorded outside, on mobile network, where my server on 192.168.x.x was not available. Then after I connected to my wifi, app uploaded all data. Maybe HTTP requests are retried?

BTW, it's not clear in UI if application loses data. I think UI should indicate it.

schollz, 2016-04-19:
Ah okay! I misunderstood.

So the app is written in Cordova and uses the jQuery ajax call to submit data. It seems that [jQuery ajax has no timeout by default](https://stackoverflow.com/questions/4148830/what-is-jquerys-ajax-default-timeout-value) and the set by the browser. I have no idea what the timeout is, but I think you are right - I guess the HTTP requests are retried until they make it to the server since it could be a long timeout.

The data is lost if you see "Error: X" instead of "Inserted Y...." as there is [a callback function for ajax related errors](https://github.com/schollz/find/blob/android/main.js#L197-L201).  

I can certainly make this more obvious on the UI. I think setting the timeout will help and then alerting the user/turning off tracking when data are lost.