schollz/find #53: Different name for the server binary

gabeguz, 2016-04-19:
`find` is the name of an ubiquitous and well known unix command.  Would you ever consider renaming the resulting binary to something else in order to avoid conflicts and confusion?

schollz, 2016-04-19:
Yes. I want to do this badly. The trouble is choosing a name. Lets open a discussion here to help rename.

Note, please [check possible names for Trademark infringement here](http://tmsearch.uspto.gov/bin/gate.exe?f=login&p_lang=english&p_d=trmk) before you suggest. There are many great names already taken :(

gabeguz, 2016-04-19:
I think the "find" name itself is fine, but perhaps the server binary could be called: 

`findd` (find daemon) or 
`find-server`?

schollz, 2016-04-19:
That's good. Its easy to change the name when building the binary. However the repo name change might have some benefits:

- You could install via `go get github.com/schollz/newname`. Right now you have a path collision like you mentioned and mentioned here: https://github.com/schollz/find/issues/32.
- The current name is a bit hard to search for (SEO)
- The current name has an inaccurate connotation for most people, since "find" is typically associated with a program that finds files