schollz/find #56: Problem trying to start find as systemd service

Christoph-Wagner, 2016-04-20:
Happens both when I compile from source or using the updated Binary.

System: Raspberry PI running raspbian jessie
go version go1.5.1 linux/arm

It works fine manually starting both.

Error log from the binary:
```
systemd[1]: Starting Find Server...                                                     
systemd[1]: Started Find Server.                                                        
find[16740]: panic: html/template: pattern matches no files: `templates/*`              
find[16740]: goroutine 1 [running]:                                                     
find[16740]: panic(0x4862e8, 0x10963000)                                                
find[16740]: C:/Go/src/runtime/panic.go:464 +0x330                                      
find[16740]: html/template.Must(0x0, 0x76e69000, 0x10963000, 0x76e69000)                
find[16740]: C:/Go/src/html/template/template.go:340 +0x4c                              
find[16740]: github.com/gin-gonic/gin.(*Engine).LoadHTMLGlob(0x10a25030, 0x520610, 0xb) 
find[16740]: C:/Users/ZNS/work/src/github.com/gin-gonic/gin/gin.go:129 +0x148           
find[16740]: main.main()                                                                
find[16740]: C:/cygwin64/home/ZNS/find2/server.go:71 +0x280                             
systemd[1]: find.service: main process exited, code=exited, status=2/INVALIDARGUMENT    
systemd[1]: Unit find.service entered failed state.                                     
```

My find.service file:
```
[Unit]
Description=Find Server
After=network.target

[Service]
Type=simple
User=root
ExecStart=/root/find/find
WorkingDirectory=/root/find

[Install]
WantedBy=multi-user.target
```

Tried it without the `WorkingDirectory=` part, made no difference.

schollz, 2016-04-20:
I think this is because I don't actually use the working directories to set the template and static files. That was a bug, and it is fixed now (hopefully) with this: https://github.com/schollz/find/commit/ffd983fd3522fdaf6a97c017c00a996a946fe2e5.

Can you try it and let me know if that works? I'm not familiar with systemd (I still do cron+/etc/init.d/ scripts).

Christoph-Wagner, 2016-04-20:
Working! :+1:

closed 2016-04-20 by Christoph-Wagner