schollz/find #60: Tracking only sends new information very slowly when screen is off on Android

Christoph-Wagner, 2016-04-20:
I'm guessing that's some wakelock problem (or rather lack of one). It would definitely be useful to have it send requests and the same speed as with screen on. Maybe with an optional timeout (default 5 minutes) to avoid draining the battery.

schollz, 2016-04-20:
I'm aware of this issue. You're right, its not great. I did some tests on this back in November (see below). Basically the phone seems to wake up for a few seconds and it spits off some fingerprints, but then sleeps for a sporadic amount of time (ranging from 25 to 120 seconds, see below).

A solution might be to use [a background plugin](https://www.npmjs.com/package/cordova-plugin-background-app) that basically would work kind of like [this alarm clock](https://github.com/MobileChromeApps/cordova-plugin-chrome-apps-alarms) except instead of alerting anyone, it would just wake and send off a fingerprint.

Another solution would be to implement wakelocks, but I've read this can take a lot of battery which is what this whole system is designed to avoid! I don't know though because I haven't tried it.

[image]
[image]

Christoph-Wagner, 2016-04-20:
I don't think there is a way to avoid battery drain when tracking with somewhat high accuracy, be it via wakelocks or via the background-plugin (however that works). That's why I mentioned a timeout. I don't know much about android internals, but my thinking would be kinda like this:  
When tracking while screen off, grab wakelock. After TIMEOUT passed, release wakelock.

schollz, 2016-04-20:
I think that's definitely worth a try. Its certainly the easiest - as [its only one plugin + two lines of code](https://www.npmjs.com/package/cordova-plugin-powermanagement). I'll play with this.

schollz, 2016-04-20:
I've added this now here: https://github.com/schollz/find/commit/6547012462a6a2ca929592161bf163f1c4b69ff5. The reuslts look good. When I use the wakelock, it works as intended (the points are the timestamps recieved of /learn requests from the phone):

[image]

versus no wakelock (previous version):

[image]

Before it becomes a release, I'd like to add an option to turn this functionality on (by default it should be off, I don't want to waste people's batteries). The without wakelock is not that bad for many applications, because of the background plugin that holds over data until the phone wakes a bit from sleep.

Christoph-Wagner, 2016-04-21:
Nice :)

schollz, 2016-04-21:
Unfortunately I don't think the wakelock is a solution. It uses a LOT of battery. I extrapolated from battery logging that it would use about ~60% per hour on my phone, even when I dimmed the screen as much as I could. For comparison, the current program, which can run in the background albeit sporadically, uses about 6% per hour (which may be the baseline of my phone).

Another possibly, yet hacky, solution is to set a silent alarm that wakes the phone up every minute or so. Then, I assume, the android might take a little bit to go back to sleep which could help with the consistency. It actually seems to work okay:

[image]

[image]

The actual implementation here could be smarter - i.e. I could create an alarm whenever a user is tracking/learning and a response from the server hasn't been seen in 30+ seconds, instead of creating alarms every 61 seconds.

Here's a direct comparison. The alarm is not as perfect as the wakelock, but it helps keep things more consistent. The only problem is that it vibrates the first time the alarm activate :|.

[image]

Christoph-Wagner, 2016-04-21:
Ouch, that's a lot ;) Well, I guess for my use case I could change it so Tasker not only starts the tracking but also wakes up the phone.

schollz, 2016-04-21:
I got a new version (https://github.com/schollz/find/commit/4b89410184677676dba3b076afeee7d629412104) that doesn't vibrate! And it also seems better than the baseline:

[image]

dextreem, 2017-12-16:
Hi all,
are there any news with this? In my case I'd like to track users taking stairs. Therefore, I need to track every 3 seconds or so. I also experienced a heavy battery drain when preventing the device from sleeping. Hence, this is not a real solution. I assume waking up the device every 3 seconds would lead to the same result?
Thanks and best regards!