schollz/find #61: Server suddenly forgot location until learn was used once or priors recalculated

Christoph-Wagner, 2016-04-20:
See the timestamps. 14:17 everything was fine, a few tracks worked when I started tracking again 22:20:03 but at 22:20:06 the server suddenly stopped knowing about the location "living room".

A get to `/location` gave

```json
{
  "time": "2016-04-20 22:22:55.197597089 +0200 CEST",
  "location": "",
  "bayes": {}
}
```

After I set the client to "learn" for 30 seconds and recalculated priors, everything was fine again. As I did both I can't say which one helped.

```
DEBUG: 2016/04/20 14:17:10 fingerprint.go:146: Tracking fingerprint for cwagner (home) at living room (guess)
[GIN] 2016/04/20 - 14:17:10 | 200 |    7.306876ms | 192.168.1.30 |   POST    /track 
DEBUG: 2016/04/20 22:20:03 fingerprint.go:146: Tracking fingerprint for cwagner (home) at living room (guess)
[GIN] 2016/04/20 - 22:20:03 | 200 |  312.958369ms | 192.168.1.30 |   POST    /track 
DEBUG: 2016/04/20 22:20:04 fingerprint.go:146: Tracking fingerprint for cwagner (home) at living room (guess)
[GIN] 2016/04/20 - 22:20:04 | 200 |  363.595676ms | 192.168.1.30 |   POST    /track 
DEBUG: 2016/04/20 22:20:06 fingerprint.go:146: Tracking fingerprint for cwagner (home) at living room (guess)
[GIN] 2016/04/20 - 22:20:06 | 200 |    7.011619ms | 192.168.1.30 |   POST    /track 
WARN : 2016/04/20 22:20:09 posterior.go:18: Not in network 
DEBUG: 2016/04/20 22:20:09 posterior.go:19: none false map[] {home cwagner tracking []} 
DEBUG: 2016/04/20 22:20:09 fingerprint.go:146: Tracking fingerprint for cwagner (home) at  (guess) 
```

schollz, 2016-04-20:
How persistence was this? After 22:20:09, did it it happen every time without knowing the location?

The /track request at 22:20:06 maybe didn't contain a whole fingerprint because it claims its `Not in network` which would mean that none of the mac addresses in the fingerprint matched any of the mac addresses that it had learned. This could happen if somehow the device sent an empty list of fingerprints.

Christoph-Wagner, 2016-04-20:
It happened continuously until I recalculated priors (rechecked the logs, it started reporting the location before I relearned, just recalculating was enough), from 22:20:09 till 22:24:09

schollz, 2016-04-20:
Haha, well I'm stumped then. I'm not sure, I'm glad its working again. I'll be on the lookout for this bug, please let me know if it happens again. :)