schollz/find #64: /learning response change

sjsafranek, 2016-04-22:
it would be nice to get a json response from learning like the following 
{
"message": "blah",
"details": {"name":"blah", "APS": 10, "user":"me!"},
"success": true
}

this would make it easier to read with programs utilizing the find system

schollz, 2016-04-22:
Whats "name"? 
Also why do you need "user"? User should be given to the server, so the client will probably know about it.

sjsafranek, 2016-04-22:
The Username is in the current message that is returned (it shouldn't be needed). Name is location name.

schollz, 2016-04-22:
Hmm. So the purpose of "message" is to relay the Debug message that the server itself into its own log. This is useful to me, as a developer, cause I can use a client to "see" the server Debug messages, since currently the App prints out the data['message'].

For practical functionality though - does the client need any of this? It already knows the username and the location and the number of APs. The client doesn't even have to output a 'message', it only does right now because its useful for my debugging purposes. Maybe the client only needs to respond to success (which should probably change to an HTTP status code).

sjsafranek, 2016-04-22:
The client probably doesn't need to see any of this. However I imagine for writing API tests it would easier to return the location name, and user within a more "machine" friendly message. If an incorrect user is returned this could highlight major async bugs that could potentially occur.