schollz/find #66: how to integrate

sander76, 2016-04-22:
Just ran into this. Looks absolutely amazing !
But I can't find anything in the info on how to integrate with other systems
like homeassistant.io or mqtt .
Can you give some pointers ?

EliasGabrielsson, 2016-04-22:
The best document I found so far is this: 
https://github.com/schollz/find/blob/docs/docs/index.md

sander76, 2016-04-22:
thanks ! That was what I was looking for....

closed 2016-04-22 by sander76

schollz, 2016-04-22:
I've never seen homeassistant.io - it looks really cool. 

The API is mostly frozen. For your purposes you likely need to poll [`/location`](https://github.com/schollz/find/blob/docs/docs/index.md#get-locationgroupxusery-or-locationgroupx-or-locationgroupxuseryzw) with a Python script and do what you want with the response. 

Please let me know if you make a little homeassistant script, itd be great to share! Also, please let me know if you need some functionality that’s not currently present on the API and I can build it in.

sander76, 2016-04-22:
@schollz 
I posted about Find on the homeassistant forum:
Seemed at least one person already knew about it ;-)

https://community.home-assistant.io/t/anyone-seen-this-find-internal-positioning/772

schollz, 2016-04-22:
Oh awesome! Except that route doesn't exist now lol. Its been replaced by simply `/location?group=X`

EliasGabrielsson, 2016-04-22:
@schollz 
FYI: There is an ongoing thread for openHAB how to utilize this framework as well.
https://community.openhab.org/t/find-high-precision-indoor-positioning-framework-for-most-wifi-enabled-devices/9909/8

schollz, 2016-04-22:
@EliasGabrielsson Thanks, I've updated their conversations with the new API as well. Sorry for confusing everyone ;)

EliasGabrielsson, 2016-04-22:
No worries, thats the nature of open source development.
Thx for the work you done, I am really impressed by the quality of the classification.