schollz/find #71: See if bluetooth / BLE can be used for find

SEJeff, 2016-04-24:
There are some small devices that would be nice to use with find which only support something like BLE (bluetooth low energy) in addition to any IOS device. Would it be possible to write something similar using bluetooth signals in place of 802.11?

schollz, 2016-04-24:
I'm going to dissect your comment into several questions, as it resonates with a couple things I've been thinking about lately, so sorry if I seem pedantic :smile: 

> Would it be possible to write something similar using bluetooth signals?

Yes, definitely. In fact, I can add Bluetooth support into the next iteration of the App using [this](https://github.com/don/cordova-plugin-ble-central). Its actually very easy to add Bluetooth because, just like WiFi routers, Bluetooth devices give off a canonical RSSI values as well as a unique MAC address.  Using Bluetooth + 802.11 would improve the classification. 

 > Would it be possible to write something similar using bluetooth signals ...*in place of 802.11?*

No, probably not. In my apartment, I see >25 access points (APs) for WiFi, but only 1-2 Bluetooth devices. Thus, unless you are an avid Bluetooth enthusiast, there might not be enough Bluetooth APs to make this useful. However, I've heard that Bluetooth is cheap, so it might be easier to buy more APs.

> There are some small devices that would be nice to use with find

Yes! I'm hoping to make some of these devices. Now that FIND is stabilizing, I've been working on designing small, cheap, open-hardware solutions for other devices that work with FIND. In theory a tracking device is very simple - just a LiPo battery with a esp8266 chip. I've ordered the parts and I'm looking forward to putting them together.

> ...in addition to any IOS device

Sorry, I don't think iOS is good option. I think some people are working on implementing a solution for iOS devices, but its not easy because it requires using [Private APIs](https://github.com/schollz/find/blob/master/FAQ.md#can-i-use-an-iphone) which Apple tries to discourage :(.

SEJeff, 2016-04-24:
WRT the private APIs, I believe you can [request the entitlement](https://forums.developer.apple.com/thread/9015) to get official access. It might be worth a shot. Even if they say no, you're not any worse off. BTW, I'm a hobbyist contributor to Home Assistant and found this project by way of a reddit comment showing integrating the two. Really cool stuff, please do keep working on it!

fermuch, 2016-11-11:
Just to add my personal use case to the discussion:

My idea is to deploy BLE (bluetooth low energy) beacons like [estimote](http://estimote.com) or [sensoro](https://www.sensoro.com/en/whyus) in a building, adding more sensors where more sensibility is needed.
Since most BLE beacons batteries last between 2 and 4 years, and don't need a power source nor any cables, they make a perfect use case for indoor location tracking; even if you don't have a lot of wifi antennas, you can still use bluetooth as a replacement and put it in places where a wifi AP don't make sense, like in a corner between two walls.

> No, probably not. In my apartment, I see >25 access points (APs) for WiFi, but only 1-2 Bluetooth devices. Thus, unless you are an avid Bluetooth enthusiast, there might not be enough Bluetooth APs [...]

With my use case, it would make a lot of sense to go with bluetooth instead of wifi by default, since it'll provide a lot more accuracy.

> Yes! I'm hoping to make some of these devices. Now that FIND is stabilizing, I've been working on designing small, cheap, open-hardware solutions for other devices that work with FIND. In theory a [...]

I'm curious why you're going with a ESP8266 instead of a BLE beacon. Or are you thinking more of a client instead of an antenna?

--

Anyways as far as I understand, this is a change for the client application and not the server, since the server only needs signal level and the antenna BSSID.

uncleashi, 2016-12-29:
Hi @schollz ,

Any update on this? Would be interesting to have this feature into FIND.

schollz, 2016-12-29:
As @fermuch says, this is only a change for the client application. The FIND server will be able to handle the signal levels (as long as they are either in the range of -100 to 0 or 0 to 100, if not its a simple fix to allow). 

Thus, in order for FIND to support this, we need to write a client application that can scan bluetooth networks and upload the bluetooth fingerprint to FIND. 

Unfortunately, I won't be able to do this. I don't have any bluetooth devices that I could readily use to develop/test this type of client application. And I don't really plan on getting bluetooth devices, since I'm happy with WiFi.  However, if I happen to come across some really cheap bluetooth senors I would absolutely give it a shot! 

That being said, I would still love to see applications for bluetooth so I would encourage someone else to take a stab at this.

uncleashi, 2016-12-30:
Hi @schollz ,

Noted. Actually, this is what I was planning to do. To make a simple app to test accuracy and performance of FIND server with BLE devices and beacons. I have quite a few BLE devices with me to test.
What is your thought on this, should we add this as a feature to FIND?
Let me know your thoughts. Thanks

schollz, 2016-12-30:
@uncleashi 

Yes, I think it would be great. 

Maybe it can be builtin to the new app?

uncleashi, 2016-12-31:
Sounds good to me.. We can add a new section into FIND app to support BLE scanning. That'll be cool feature to have in it. Let me give it try. :)

fotiDim, 2017-12-20:
Has there been any progress on that? I think it would work great for pet collars. I wouldn't want to use a wifi chip there.