schollz/find #75: Better coding documentation

schollz, 2016-04-25:


candaj, 2017-01-16:
Hello,

First, I want to say that's an amazing project you have  ! I tested it yesterday locally and it worked almost perfectly ! (just a little bit slow on real-time tracking when i change of room, something to do with not a lot of learning phase i think). 
I would love some more technical documentation in order to see how it's works and maybe contribute on it :)

Best luck !

Julien

schollz, 2017-01-16:
Sure! Is there anything in particular you'd like to contribute to? I can prioritize the documentation.

candaj, 2017-01-16:
Nothing in particular for now. I need to be more familiar with go since I just start with it. 
But I'll watch your repo closely and propose PR if I've some ideas :)