schollz/find #82: MQTT support

wutu, 2016-04-28:
Great software, you plan to support MQTT? It would certainly facilitated the integration with multiple systems.

schollz, 2016-04-28:
Yes, I think this would be a great addition. I don't know much about it. I'll open this up as a feature request for others to help with!

Seems like there are some Golang packages to get started with: https://github.com/yosssi/gmq

schollz, 2016-04-29:
Okay, it was a lot easier then I thought! I added it already. Its not live yet as I'm not sure what kind of info you'd like. Should I just pass the JSON of the current location data through the MQTT messages?

wutu, 2016-04-29:
superfast, thx.

tavalin, 2016-04-29:
Is it possible to specify a private MQTT server?

schollz, 2016-04-30:
I've now realized that MQTT is in no way private by default lol. I'm taking it offline for now.

I think a "okay" solution will be to allow users to specifically request a MQTT connection, at which I will add their Group to a ACL file that allows access. Of course, still anyone can access this, but this is the general scheme of how FIND currently works (security through obfuscation). The main thing here is that I need to disallow the wildcard rooms so that people can't spectate all the transactions...

I'm open to other suggestions too!

schollz, 2016-04-30:
Okay I found a way to do this, its going to be limited to Linux systems for now.

schollz, 2016-04-30:
Okay, sorry, I have a "secure" MQTT now (https://github.com/schollz/find/commit/d34af10964bf48619d6fd674ebee676ca1ecf9a8). Basically you have to subscribe to have MQTT access, and then you can authenticate yourself in the MQTT channel so others can't snoop.

[Read this to get started](https://github.com/schollz/find/blob/docs/docs/mqtt.md) and let me know if you have questions!

closed 2016-04-30 by schollz

tavalin, 2016-04-30:
@schollz Did you have any comments on allowing users to specify the mqtt server?

schollz, 2016-04-30:
@tavalin Do you mean, allowing users to specify a different mqtt server that you can listen to the locations determined from FIND? I.e. having FIND forward the messages to a different MQTT server? This doesn't exist, but I can add it if you need that.

tavalin, 2016-04-30:
Yes. As I'm running my own Find server I'd also like Find to forward the location info to my own MQTT server too (rather than have my data in the cloud).

schollz, 2016-04-30:
Okay, that you can do. [Here are the instructions](https://github.com/schollz/find/blob/docs/docs/mqtt.md). Right now its only setup for mosquitto. Let me know if you have problems.