schollz/find #83: Introduce privacy to FIND

abcdev, 2016-04-29:
There is no real concept of privacy in FIND yet. Therefore I suggest some changes to the way FIND works.

1. Change the ML to a supervised, offline classifier (eg SVM or NN) (opencv / tensorflow)
2. Send the latest successfully trained model to the clients
3. Clients classify their fingerprints on their own

schollz, 2016-04-29:
You are right. HTTPS is in place, but really any current privacy is through obfuscation - basically your "secure" as long as someone can't guess your Group name.

So currently the model is Naive-Bayes which could be used the way your suggesting. The output of the current model is basically a lookup table which then is used with a fingerprint with a very simple calculation. The lookup table itself stays modestly small (300 kb - 2 mb) so it could easily be zipped and sent downstream to the client. I have tried NN, and they are just as good (if not worse) as NB. I haven't tried SVM. I think that this could be built in for clients that want to do the computation internally (i.e. smartphones / laptops but not microcontrollers). I will add this to the milestones for 3.0.

As for privacy in general, we are working on two other microservices - a GIS server and a Authentication server that would provide full-form privacy and security (through API keys, authentication requirements and encryption of location names). This is not ready quite yet though.

schollz, 2016-05-01:
I've introduced a Privacy Policy (https://github.com/schollz/find/commit/3b9c914de963a3bbe77eff32cb0d08746c21fe0c) with a disclaimer (https://github.com/schollz/find/commit/f27e47b9975d978d7f92ddf75141ef08ac2cffbf) about the current security of FIND: https://doc.internalpositioning.com/privacy/ which is now prominently displayed in the main window.

The milestones have also been updated with the new Authentication and GIS components.

closed 2016-05-01 by schollz

abcdev, 2016-05-02:
My main concern is the dependency on a server telling the clients where they are and therefore is knowing their position/tracking them. The size of pretrained SVM models scale rather well in this domain and the classification is cheap in aspects of computation.(To be taken with a grain of salt, because I was in this workgroup) I'd also like to quote some findings from the Paper "[Practical Concerns of Implementing Machine Learning Algorithms for W-LAN Location Fingerprinting](https://www.informatik.fb2.frankfurt-university.de/~jschaefer/assets/Preprint-FP-01s.pdf)" :

 4) The  Naive  Bayesian  approach  fails  because  the assumptions  of  independence  (and  normal distributions) are violated.
 5) However,  if  a  PCA*  is  performed,  Bayesian  performs  quite  well  and  is  only  slightly  worse than SVM and KNN**.

*PCA (principal component analysis).
**KNN (k-nearest neighbor)

schollz, 2016-05-02:
Cool paper! Thanks for the link. I'd like to try SVM.

As for NB, why is the assumption of independence and normal distributions violated?

For the distributions, it's true that sometimes you see bimodal distributions, but most data stays pretty normally distributed (this is Gaussian filtered) so it doesn't seem terribly violated:

[image]

As for independence, the probability of a router giving a certain signal given you're in a location, P(routerX = signalX | locationY) seems to me to be independent. The routers X=1..N are all in different places and, unless their RF signals interfere, I don't see how can they be correlated.

schollz, 2016-05-03:
@abcdev You inspired me to add SVM to FIND - so far it works pretty good: https://github.com/schollz/find/tree/svmcalc