schollz/find #95: Make MQTT built-in

schollz, 2016-05-14:
Possible pure-Go solutions:
- https://github.com/surgemq/surgemq
- https://github.com/jeffallen/mqtt

johnjore, 2016-05-20:
Or optional? I already have mosquitto running for OwnTracks. No need to have two of these running...

schollz, 2016-05-20:
Definitely, I will configure the server to be optional.

lwis, 2016-07-07:
@schollz would be cool if it could emit MQTT data, also.

JackTheDipper, 2016-11-16:
@johnjore, @lwis
Does that mean you would be happy if #121 gets implemented or do you prefer another solution?

lwis, 2016-11-16:
@JackTheDipper I'm not exactly clear on what #121 is proposing, sorry.

JackTheDipper, 2016-11-16:
@lwis, heh, ok.
It is supposed to mean that findserver should (optionally) connect to external MQTT brokers so that you can let it post updates to a mosquitto server that you are already running. Currently, you would have to start another mosquitto instance handled/configured by findserver in such a case.

I thought "emit MQTT data" would probably mean "post updates to an MQTT broker", but maybe I 'm wrong on this. ;-)

lwis, 2016-11-16:
@JackTheDipper yeah, that's what my request was for, we're on the same page :)

lwis, 2016-12-08:
@JackTheDipper did this progress? I would love to be able to stop polling the FIND server for a users location and just have it immediately broadcast on my mqtt server instead.