schollz/find #96: Add timestamp to mqtt payload so that it matches /location

tavalin, 2016-05-14:


schollz, 2016-05-14:
Fixed: https://github.com/schollz/find/commit/d887f336a242f3747335a7f133a414662d68951c.

Its in release 2.1 (updated) and live on the server as well.

tavalin, 2016-05-14:
Just downloaded https://github.com/schollz/find/releases/download/v2.1/find-2.1-linux-arm.zip and I'm not getting timestamp in the MQTT payload. It's still like so:

`{"location":"bedroom","bayes":{"bedroom":1},"svm":{}}`

schollz, 2016-05-14:
Weird lol. I just tested the linux-amd64 build version and its working fine. I don't have an environment setup to test on arm version right now, but I can try later tonight. 
In the happenchance that I didn't upload the newer arm version, I just deleted the old arm and rebuilt and reuploaded. So try again? :smile: 

Also...you can always build from source... it's easy! Here's some instructions to install Go and install FIND:

## Install Go

Download and install:
```
wget https://storage.googleapis.com/golang/go1.6.2.linux-armv6l.tar.gz
sudo tar -C /usr/local -xzf go1.6.2.linux-armv6l.tar.gz
```

Add a `work` directory to store Go dependencies.
```
mkdir ~/work
```

Then `nano ~/.profile` to add these lines:

```
export PATH=$PATH:/usr/local/go/bin
export GOPATH=$HOME/work
```

Reload the profile, and that's it! Test with 

```
-> % source ~/.profile
-> % go version
go version go1.6.2 linux/arm
```

## Install FIND

```bash
git clone https://github.com/schollz/find.git
cd find
go get ./...
go build
```

Then you have the binary `find` you can just run.

schollz, 2016-05-15:
Okay, sorry I can't replicate your error. Are you sure you downloaded the latest version?

This is what I'm getting from listening to my mosquitto server using the `testdb` and `mosquitto_sub -h 127.0.0.1 -u testdb -P QnWCbm -t "testdb/location/#"`:

```javascript
{  
   "location":"zakhome floor 2 office",
   "time":1463274067344066860,
   "bayes":{  
      "zakhome floor 1 kitchen":-0.3331607643545871,
      "zakhome floor 2 bedroom":-0.7296046268131593,
      "zakhome floor 2 office":1.0627653911677468
   },
   "svm":{  
      "zakhome floor 1 kitchen":-6.65085205272294,
      "zakhome floor 2 bedroom":-5.906190010886762,
      "zakhome floor 2 office":-0.004023081751909232
   }
}
```

As you can see the time is there!

tavalin, 2016-05-15:
Using the new v2.1b works so close this :smile:

closed 2016-05-15 by tavalin