schollz/find #97: Problems running FIND app in background

schollz, 2016-05-14:
> Samsung S7 Edge running Android 6.0.1. Find App is V0.5. If I enter tracking mode then press the power button to lock the screen tracking continues to work fine. If I press the Home button first the behavior is different and I get those /track posts without the location guess.

Original: https://community.openhab.org/t/find-high-precision-indoor-positioning-framework-for-most-wifi-enabled-devices/9909/43?u=schollz

sjcliffe, 2016-08-01:
Just confirming this bug - I have an S7E running 6.0.1 too and see the same issue.

holgerdunst, 2016-08-30:
I don't know if it is the same probleme but I noticed that running the app in background the value "Last seen" on the dashboard at https://ml.internalpositioning.com/dashboard/  isn't updated anymore. I am using android  6.0

ihightower, 2016-11-03:
Hi I am confirming this bug also.. The app doesn't update location if it goes in background.

However, if I just power off (using the power button).. it continues to update the location.
If switch ON again.. and changing to other apps.. it doesn't update again.

Pls fix this.

OS: Android 6.0.1, OxygenOS version 3.1.0
Model: Oneplus Two, ONE A2003

Also Same behaviour Samsung Galaxy J7 (2016) SM-J710FN, Android 6.0.1

schollz, 2016-11-03:
Thanks! 
I wonder, could you try the 3rd party native app [Room+](https://play.google.com/store/apps/details?id=com.seversion.roomplus&hl=en) and see if that works better?

ihightower, 2016-11-07:
Thank you.. Room+ App seems to work running the background.

But I noticed different behaviour when the app is running the background... (or may be it is just me I imagined things as I can't reproduce the behaviour)...

Here are the different behavious:
1) the app doesn't update while in background... (i couldn't reproduce the problem again, but it did happen as far as i can remember)
2) it updates every 3 to 7 seconds or about.. (majority of the cases).
3) it updates only every 2 minutes (observed by my colleague from the server connections).

Any guidance on it will be appreciated.

A separate question...  how can one implement a solution where use GPS data when available and use IPS (indoor positioning.. which is FIND).. when GPS is not available, etc. Any ideas? Is there an app that you think can do both.

Thank you so much.

schollz, 2016-11-07:
@ihightower Glad its working kinda. :)

> 1) the app doesn't update while in background... 

I'm not sure why, this might be a Room+ problem or an Android 6.0.1 problem.

> 2/3) it updates every 3 to 7 seconds or about

The issue with the spurious updating is not a new one - [there is a discussion about it here with data similar to what you are experiencing](https://github.com/schollz/find/issues/60) (updates 3-7 seconds  and sometimes up to 2 minutes). I think it is controlled by how the Android phone "sleeps" when it is not in use. The only way around this was to use a wakelock, but thats a terrible solution because it keeps the screen on and uses a ton of battery. I'm not sure a way around this yet.

> how can one implement a solution where use GPS data when available and use IPS (indoor positioning.. which is FIND)

I've been thinking about integrating GPS data, but as of yet there is no way to do this yet.

ihightower, 2016-11-09:
@schollz Thank you so much for your comments. Very helpful.