schollz/find #99: Unknown Location

johnjore, 2016-05-21:
I was planning on using 'find' in OpenHAB to detect house mode "Sleep" (I have three modes; Home, Away, Sleep). I used the learning mode for my office and bedroom on the 2nd floor and it looks to work reasonably well. The idea being if the location remains "Bedroom" for 15? minutes, then house mode should change to "Sleep".

However, when moving to the ground floor, it still reports either "office" or "bedroom", instead of "unknown".

How can 'find' be configured to report "unknown" when the margin of error is large enough to not really be any of the learned locations? I was hoping to NOT need to use learn on every floor/room in the house? If my approach is wrong, what is the correct approach to achive my goal?

JJ

schollz, 2016-05-21:
> How can 'find' be configured to report "unknown" when the margin of error is large enough to not really be any of the learned locations?

You can't. As long as the the fingerprint contains a known access point, the server will try its best to figure out which location it is (among the learned locations). There are a couple reasons for this. One is that the Bayes metric is not super quantitative and might vary between environments so its hard to determine a cutoff/threshold to set. I will be experimenting with this in the future.

For the present, I suggest learning your ground floor as a separate location. You don't need to learn every room on the ground floor - you can simply set learn to "ground floor" and then walk around your entire ground floor. That way the server will know that entire location as one "room."

johnjore, 2016-05-22:
Thanks for the clarification!

closed 2016-05-22 by johnjore