schollz/find #103: Allow config file to be used in place of runtime options

schollz, 2016-07-12:
Especially useful for [Docker, e.g. issue 38](https://github.com/schollz/find/pull/38).

tavalin, 2017-01-06:
Hi @schollz, has there been any progress on this? I just installed the docker version and ran into the issue that there's no config file so I'm unable to set the runtime options e.g. ports, MQTT broker IP address

schollz, 2017-01-06:
Hi @tavalin , I should be able to do this over the weekend.

The main trouble is how to get the PID of `mosquitto`, which I can resolve just by using the `mosquitto` PID file.

In the meantime, the workaround is to simply clone find and then build the Docker image yourself after changing supervisor.conf to your liking.

tavalin, 2017-01-06:
Incidently I've already got an MQTT server on my network, so having Find just run as an MQTT client (instead of a server) is fine with me :)

schollz, 2017-01-06:
What is the preferred method of sending the configuration to the inside of the Docker container? Is it acceptable to just map configuration file directly, e.g. `docker run -v /home/zack/find.conf:/etc/find.conf ....`?

tavalin, 2017-01-07:
I'm not a Docker expert but that's what I'm doing for the other 2 containers I'm running. I can then backup just the configuration folders and in the event of an issue I can recreate the image with the config and be back up and running again.

schollz, 2017-01-07:
I think its working, you might try your setup with this branch: https://github.com/schollz/find/tree/dev

tavalin, 2017-01-08:
I've had a bit of time to test it.  Everything seems OK when run on my Ubuntu box but I'm having an issue with MQTT when using Docker.