schollz/find #104: Improving FIND for Docker

schollz, 2016-07-16:


schollz, 2016-07-16:
I've complicated the Dockerfile a bit now.

It should be able to include `mosquitto` and `SVM` in the box, and currently it does. Its based off the Ubuntu16 image, but that could be changed (except `mosquitto` is not bundled in earlier versions).

Thom-x, 2016-07-16:
Looks good to me except the entry point.
I thkink we should only run the application at this point, not build it and run multiple things at a time.

Take a look at this Dockerfile, it use [supervisord ](https://github.com/Thom-x/docker-fr24feed-piaware-dump1090-mutability/blob/master/supervisord.conf) and [tini ](https://github.com/krallin/tini/) to run the container :
https://github.com/Thom-x/docker-fr24feed-piaware-dump1090-mutability/blob/master/Dockerfile

schollz, 2016-07-16:
@Thom-x , Thanks, that makes a lot more sense. I've updated it (https://github.com/schollz/find/commit/72c32bb9161a8763227efc324e079aa8eb4fb37d).

closed 2016-08-02 by schollz