schollz/find #108: support for iOS devices with iOS 10?

Bart274, 2016-08-02:
Hi,

Do you think it's possible to get this working with iOS 10 now?

schollz, 2016-08-02:
Good news, apparently [it is available in iOS 9+](http://stackoverflow.com/questions/32970711/is-it-possible-to-get-wifi-signal-strength-in-ios-9/32971064#32971064).

I would need help to develop this since I've never programmed for iOS devices.

Bart274, 2016-08-02:
I'm glad to see it should work for iOS devices now as well! :)
Unfortunately, I can't help with the development as I have no experience for developing for iOS :(

abcdev, 2016-08-19:
I have requested a hotspot helper network extension in early august but have received response.

schollz, 2016-08-21:
Do you know how long it usually takes?

abcdev, 2016-08-23:
Nope, but my request dates back almost three weeks. It seems quite long even for apple standards. Maybe they don't reply when they reject a request ?

jonathanweinberg, 2016-09-03:
I'm not shocked that are running wickedly behind since there is steady rumors of not only the iPhone refresh, but several other platforms being refreshed alongside (MacBook\iMac + Watch).

Wish there were some way to help.

Bart274, 2016-09-06:
I really hope someone can help us out with this :)
that, or we can use bluetooth?

abcdev, 2016-09-13:
This is the response from apple I got today:

Hello,

Thank you for your interest in the Network Extension framework. Unfortunately, these APIs are not designed for the use you’ve identified. We’ll let you know if that changes.

schollz, 2016-09-13:
That's really unfortunate. I wonder if "not designed" means "not allowed"?

jonathanweinberg, 2016-09-15:
And along that same token, if you happened to do one of the "intended" purposes, while doing the things we need to do (fingerprinting sequence), it might be allowable.

jonathanweinberg, 2017-09-18:
Maybe after launch of iOS 11, another request might be possible?

schollz, 2017-09-18:
That would be fantastic. I'm not an iOS person, so if someone else can weigh in that would be helpful!

abcdev, 2017-09-24:
@schollz By now I have played around with the NEHotspotHelper API. Bad news though, it is not feasible for an indoor location service. You get no guarantee to be called when having registered a Hotspot Helper. It is triggered every time when opening the WiFi settings but thats the only "pattern" I saw. It is designed to help the user choose a WiFi an app provides further info for but that's all. See also [Apple Developer Forum](https://forums.developer.apple.com/thread/47300)