schollz/find #109: trim inputs

Dutchy-, 2016-08-04:
I noticed that inputs from the android app are not trimmed before use.
I have a keyboard which tends to add spaces after a word when using completion, so I added a group "Home ". Consequently I had to use `/location?group=home%20` to access my data. I understand how that works, but i recommend trimming input.

Dutchy-, 2016-08-04:
I also suspect some unexpected errors where some parts (e.g. the dashboard) do trim, so it thinks it has no data? I'm not sure, but should be fairly easy to reproduce.

schollz, 2016-08-05:
Thanks, I'll look into this.

closed 2016-08-08 by schollz

schollz, 2016-08-08:
@Dutchy- 

If you are using the public server and you need to migrate/rename your DB, let me know at zack@hypercubeplatforms.com. I still haven't implemented a route for automated migration, but its in the works! :)

Dutchy-, 2016-08-09:
No I'm good, I use my own server :)

commit 3a8b0ed0 (2016-08-08): Trimming values to Fix #109