schollz/find #112: problem during build

AnduriI, 2016-08-21:
Hey,

first to say: realy nice framework you build there. I tried with your only server and it's awesome. Now to realy use it I tried to install locally on my rpi3. Needed some time to figure out how, but finally got go installed and seems to be working. but during ```go build``` I get the following error:
```
~/find $ go build
warning: GOPATH set to GOROOT (/home/pi/go) has no effect
mqtt.go:23:2: cannot find package "git.eclipse.org/gitroot/paho/org.eclipse.paho.mqtt.golang.git" in any of:
        /home/pi/go/src/git.eclipse.org/gitroot/paho/org.eclipse.paho.mqtt.golang.git (from $GOROOT)
        ($GOPATH not set)
api.go:20:2: cannot find package "github.com/boltdb/bolt" in any of:
        /home/pi/go/src/github.com/boltdb/bolt (from $GOROOT)
        ($GOPATH not set)
routes.go:20:2: cannot find package "github.com/gin-gonic/contrib/sessions" in any of:
        /home/pi/go/src/github.com/gin-gonic/contrib/sessions (from $GOROOT)
        ($GOPATH not set)
api.go:21:2: cannot find package "github.com/gin-gonic/gin" in any of:
        /home/pi/go/src/github.com/gin-gonic/gin (from $GOROOT)
        ($GOPATH not set)
fingerprint_ffjson.go:11:2: cannot find package "github.com/pquerna/ffjson/fflib/v1" in any of:
        /home/pi/go/src/github.com/pquerna/ffjson/fflib/v1 (from $GOROOT)
        ($GOPATH not set)
```
Does anyone knows what the problem is? I'm not very familiar with go or linux.
Thanks a lot for your help in advanve.

Regards,
AnduriI

schollz, 2016-08-21:
You need to install the dependencies.

You should run:

	go get github.com/boltdb/bolt
	go get github.com/gin-gonic/contrib/sessions
	go get github.com/gin-gonic/gin
	go get github.com/stretchr/testify/assert
	go get github.com/pquerna/ffjson/fflib/v1
	go get git.eclipse.org/gitroot/paho/org.eclipse.paho.mqtt.golang.git

Or, you can just run

	make

In the directory where you downloaded it.

AnduriI, 2016-08-21:
Thanks for your hint. I tracked the problem down a litte bit, hopefully you can help me with that.
Seems that I messed up gopath and goroot when installing it. First my system didn't noticed go, so I tried to set these variables. Now I know that goroot has to point to my installation ($HOME/go) and gopath should be a working directory for the packages. I have both set to $HOME/go, which is of course not usable. Can you tell me what to set gopath to? Is it specific for find or global for all go-project (just in case there will be another on my rpi in future)?
Thanks a lot.

Regards,
AnduriI

schollz, 2016-08-21:
`GOPATH` is a global folder where Go will store dependencies. Its common to make a new folder, `$HOME/work` for the it.

Then just make sure your paths are set correctly (assuming you installed Go into `$HOME/go`):

```
export GOPATH=$HOME/work
export PATH=$PATH:$HOME/go/bin
```

AnduriI, 2016-08-21:
thanks a lot @schollz, that worked. After using go build I can now run ./find. But it seems a little strange (no version number) and I can not connect to it with a brower to see the graphs like in your online server. Is SVM needed, I see a few lines kind of advertising it:
```
~/find $ ./find
/home/pi/find/data
SVM is not detected.
To install:
sudo apt-get install g++
wget http://www.csie.ntu.edu.tw/~cjlin/cgi-bin/libsvm.cgi?+http://www.csie.ntu.edu.tw/~cjlin/libsvm+tar.gz
tar -xvf libsvm-*.tar.gz
cd libsvm-*
make
cp svm-scale /usr/local/bin/
cp svm-predict /usr/local/bin/
cp svm-train /usr/local/bin/
-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----
   _________  _____
  / __/  _/ |/ / _ \  ______ _____  _____ ____
 / _/_/ //    / // / (_-< -_) __/ |/ / -_) __/
/_/ /___/_/|_/____/ /___|__/_/  |___/\__/_/

(version  build devdevde) is up and running on http://192.168.38.18:8003
-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----
```

schollz, 2016-08-21:
No problem.

All appears well. No version numbers when you build yourself (its version "dev"). You don't have to install SVM, its totally optional. Honestly I don't find SVM provides much benefit, but I include it anyways just in case people are curious.

What happens when you try to connect with a browser? You should be able to see something when you goto http://192.168.38.18:8003.

AnduriI, 2016-08-22:
Well yesterday it didn't work, but today... everything fine. I can connect with a browser and already learned the first fingerprints with my android.
One last question: do you also have a script to enter in init.d or comparable to make find run at boot?

schollz, 2016-08-22:
Sure, this is my init.d script. Save it as `/etc/init.d/find.init` and then start it with `/etc/init.d/find.init start`. Make sure to change `dir` to the directory you have the binary. Also modify `cmd` for the options you want.

```
#!/bin/sh

### BEGIN INIT INFO
# Provides:          find
# Required-Start:    $remote_fs $syslog
# Required-Stop:     $remote_fs $syslog
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Start daemon at boot time
# Description:       Enable service provided by daemon.
### END INIT INFO

# This should be placed in /etc/init.d directory
# start with
# sudo /etc/init.d/find start
# stop with
# sudo /etc/init.d/find start
name="find"

dir="/LOCATION/TO/FINDSERVER"
user="root"
cmd="./$name"

pid_file="/var/run/$name.pid"
stdout_log="/var/log/$name.log"
stderr_log="/var/log/$name.err"

get_pid() {
    cat "$pid_file"
}

is_running() {
    [ -f "$pid_file" ] && ps `get_pid` > /dev/null 2>&1
}

case "$1" in
    start)
    if is_running; then
        echo "Already started"
    else
        echo "Starting $name"
        cd "$dir"
        sudo -u "$user" $cmd >> "$stdout_log" 2>> "$stderr_log" &
        echo $! > "$pid_file"
        if ! is_running; then
            echo "Unable to start, see $stdout_log and $stderr_log"
            exit 1
        fi
    fi
    ;;
    stop)
    if is_running; then
        echo -n "Stopping $name.."
        kill `get_pid`
        for i in {1..10}
        do
            if ! is_running; then
                break
            fi

            echo -n "."
            sleep 1
        done
        echo

        if is_running; then
            echo "Not stopped; may still be shutting down or shutdown may have f                                                            ailed"
            exit 1
        else
            echo "Stopped"
            if [ -f "$pid_file" ]; then
                rm "$pid_file"
            fi
        fi
    else
        echo "Not running"
    fi
    ;;
    restart)
    $0 stop
    if is_running; then
        echo "Unable to stop, will not attempt to start"
        exit 1
    fi
    $0 start
    ;;
    status)
    if is_running; then
        echo "Running"
    else
        echo "Stopped"
        exit 1
    fi
    ;;
    *)
    echo "Usage: $0 {start|stop|restart|status}"
    exit 1
    ;;
esac

exit 0
```

AnduriI, 2016-08-22:
You are great, thanks. Works perfectly.

closed 2016-08-22 by AnduriI

schollz, 2016-08-22:
Great, let me know if you have any other questions!