schollz/find #117: How to open the exported .db file ?

ckybonist, 2016-10-18:
I exported my fingerprint data from dashboard (ml.internalpositioning.com/dashboard/...), while don't know how to open this binary data.

Could anyone tell me which database program is suitable for opening this *.db file.

schollz, 2016-10-18:
Its a BoltDB database file, which can be inspected using the [boltd tool](https://github.com/boltdb/boltd).

Alternatively, you can use FIND to dump the learning and tracking fingerprints directly as JSON. If your file is called `MYSTUFF.db` then use:

```find -dump MYSTUFF -data /path/to/MYSTUFF.db```

This will create a directory called `dump-MYSTUFF` with two files containing all the learning and tracking fingerprints. I realize this is a little clunky, but no one has used this feature to my knowledge (except me) and I'd be happy to improve it!

ckybonist, 2016-10-18:
Thank you, I'll try it later 👍

closed 2016-11-21 by schollz