schollz/find #121: Add support for MQTT brokers that are not directly managed by FIND

JackTheDipper, 2016-11-14:
Currently, findserver creates/overwrites a mosquitto configuration everytime it is started. Hence, already-running mosquitto instances on a server cannot be used and own configuration parts cannot be appended.

Please add support for connections to MQTT (or mosquitto) brokers without forcing a find-only configuration.

(And many thanks for this very interesting piece of software!)

schollz, 2016-12-03:
Thanks, this is a good feature. 

Do you know of the best way of doing this? I'm really not so familiar with mosquitto (I familiarized myself only enough to get it working :)). For instance, does mosquitto support multiple config files?

JackTheDipper, 2016-12-03:
Would it be necessary to update the FIND part of the mosquitto configuration every now and then? If not, maybe the easiest way would be just to connect to mosquitto and let the (static) configuration be up to the user. 

Alternatively, at least in Debian, there is a conf.d directory by default in which you can put several configuration files, located in /etc/mosquitto/conf.d/. (This is implemented by a main mosquitto.conf file, which contains an "include_dir" statement.)

schollz, 2016-12-03:
@JackTheDipper the reason that the mosquitto configuration is reloaded is to update the access control list. 

I want every user to have password protection on their channel. Users are added dynamically, which means mosquitto needs to respond dynamically. The easiest way to do this I found was to rewrite the passwd file, the acl file, and then reload the configuration.

There is another way to do this - using a [DB backend plugin for mosquitto](https://github.com/jpmens/mosquitto-auth-plug). I really don't want to do this though, because it adds yet another thing that FIND would need.

I would opt for what you mention, taking advantage of the conf.d directory.

JackTheDipper, 2016-12-03:
Ah, ok. So, a dynamic configuration is indeed required.

Sounds sensible. For a distribution-agnostic way, findserver may be allowed to maintain its own configuration directory and existing mosquitto instances can include it by using "include_dir". (Debian packages might install the files directly to conf.d provided that they are uniquely named, i.e. findserver_passwd or similar). Or tell the user in README.debian that they shall add the "include_dir" statement if they want to activate find support in their default mosquitto instance. This might be the safest way.) What do you think?

sjcliffe, 2017-01-05:
I have an existing MQTT "network" that I would like to integrate find with and this update looked like it would help but I hit two problems. 

Firstly, I was running the findserver as a different user to mosquitto so it was unable to send a signal to reload the config.

After I restarted findserver as the same user it broke my MQTT network as I have a bunch of anonymous clients and findserver sets "allow_anonymous false". 

I'm going to look at modifying it to support an existing external mosquitto configuration just publishing anonymously to a topic.

sjcliffe, 2017-01-06:
It was pretty easy - a small modification to server.go and mqtt.go

Patch file attached in case anyone is interested.

[mqttexisting_patch.txt](https://github.com/schollz/find/files/688909/mqttexisting_patch.txt)

schollz, 2017-01-06:
@sjcliffe Thanks! I think your changes are good, would you mind sending a PR with that?

sjcliffe, 2017-01-08:
PR submitted @schollz

JackTheDipper, 2017-07-17:
Thank you very much for this addition, @sjcliffe and @schollz! This greatly solves this issue.

Closing.

closed 2017-07-17 by JackTheDipper