schollz/find #124: Android client app update..

uncleashi, 2016-11-24:
Hi,

I was going through the project. Managed to build Android client app but found that the app on Google Play store and the current git app are different. The app on github was last updated 7 months ago.
Can you please update it?

schollz, 2016-12-03:
I'd like to, but I don't have the time to right now and its not the focus of this project. I'd suggest trying [Room+](https://play.google.com/store/apps/details?id=com.seversion.roomplus&hl=en), which is a 3rd party app that is kept up to date.

The app in this repo is certainly functional, but it does need an update to use the new permissions handling of Android devices. I don't really have the expertise to do this and would love help if someone wants to improve the open-source app.

uncleashi, 2016-12-07:
Hi Schollz,

Noted.
I have seen Room+ app but there are few issues which I found in it. 
Irrespective of that I would like to contribute to this project. I am more comfortable with Android development. And can port the complete logic onto Android environment. And I also think that it would be a better thing to do if we look at long term support for this project.
Let me know your views.
Thanks

schollz, 2016-12-07:
@uncleashi 
That would be wonderful!

The logic is very simple. All the app needs to do is capture the RSSI and signal strength of surrounding WiFi networks, and then format that information into the following JSON:
```javascript
{
   "group":"some group",
   "username":"some user",
   "location":"some place",
   "time":12309123,
   "wifi-fingerprint":[
      {
         "mac":"AA:AA:AA:AA:AA:AA",
         "rssi":-45
      },
      {
         "mac":"BB:BB:BB:BB:BB:BB",
         "rssi":-55
      }
   ]
}
```
This JSON is sent via `POST https://ml.internalpositioning.com/learn` or `POST https://ml.internalpositioning.com/track` depending on whether it is learning or tracking. The user, group name and the switch to learn/track should part of the UI that the user can modify. The other major feature is that it needs to be able to run in the background. The current app does this, but as you mention, the current app lags behind in the ability to set permissions according to the new Android standards.

uncleashi, 2016-12-07:
Awesome.
So, are you comfortable if I port this complete logic into Android JAVA environment, as I am more comfortable with it?

schollz, 2016-12-07:
Yep, that sounds great!

uncleashi, 2016-12-08:
Cool.
Lets get this rolling. Will share the brief with you about what all I am planning to add in the app. And then we can kickstart.

uncleashi, 2016-12-15:
Hi Schollz,

have started working on the app. I have couple of question:
#1. I was going through your android app code. In the fingerprint there is a tag "Password". Is it still needed?
If yes how does it works if someone is setting up their own server?
Following is the snippet from your code:-
var fingerprint = {
            "group": "some-group",
            "username": "some-user",
         ->   "password": "none",
            "location": currentLocation,
            "time": Date.now(),
            "wifi-fingerprint": network_data
        }

#2. What is the ideal scanning interval in case of:
a. Tracking
b. Learning
And want to understand the consequences if we increase the scanning period.

Thanks

schollz, 2016-12-15:
Great @uncleashi ! Here are some answers:

1. Passwords are *not* needed anymore. There are no passwords on the site, and I have no plans for passwords.

2. The ideal scanning interval for learning is ~3 seconds. Faster than that and phones tend to fritz out. The ideal scanning interval for tracking is maybe ~10 seconds, so that it can detect that you are in a new location relatively quickly (in 10 seconds).

3. I would send an object after every scan - the realtime applications are what I find most useful and that won't be possible with batching.

Thanks, please let me know if you have more questions!

bodiroga, 2016-12-15:
Hi @uncleashi !

It's awesome to see that you are already working on it, great news! Many thanks!

If I'm allowed, I want to comment that for question 2) I would love to see the option to change the tracking interval dynamically through an Android intent. One of the best features of the Room+ application is it's integration with Tasker, but making a specific Tasker plugin is not needed as a first step, the Android intent listener is more than enough for now. Anyway, this is not a feature that should be available for a 0.1 release, so you can forget it for now. I just wanted to point it out ;)

Keep up with the good job and best regards,

Aitor

uncleashi, 2016-12-16:
Thanks Schollz.

One more question, what location needs to be POST in case of TRACKING?

uncleashi, 2016-12-16:
Hi @bodiroga ,

Thanks for feedback.
I am planning to add this into settings intent wherein one can change Learn and Track scanning interval. :)

As far as I know Room+ allow you to change BG scan interval. Just a guess as I have not used it PRO version.

schollz, 2016-12-17:
@uncleashi 

1. The "location" is ignored when tracking, and you need not send it.
2. The time is a integer representing Epoch time (seconds since 01/01/1970). There should be a standard routine to do this.

uncleashi, 2016-12-18:
Awesome.. thanks..

uncleashi, 2016-12-18:
Hi @schollz,

Wanted to understand FIND behaviour in following cases and which will be the best case to use for learning:
There are 3 rooms next to each other. I want to map each of them on FIND. Now here are the cases:
1. I walk around the periphery of the room for 34-5 min and get all possible rssi values of all APs into learning algorithm.
2. I stand in middle of the room and train the system for that point in space only.
Which will be best case and why?
Accordingly I am planning to add few things in upcoming versions in terms of learning of algorithm. :)
Also a quick update, the version 0.1 is almost done. Planning to commit it in a 8-10 days time if all goes right in final testing. 
Thanks

schollz, 2016-12-18:
Cool, great progress!

In terms of learning modes, I haven't tested so I can't tell you for sure. I suppose that 1 is better, but I usually just do 2 with results that are just fine.

uncleashi, 2016-12-19:
Noted and thanks. :)

uncleashi, 2016-12-23:
Hi @schollz ,

I have finished the v1.0 of android client app and in a day or 2 time I can push it into repo. Let me know how can we go about pushing it.
Thanks :)

schollz, 2016-12-23:
@uncleashi That's awesome!

Actually, would you like to make your own repo e.g. github.com/uncleashi/find-app? That way you can be the maintainer since you did the coding. 

Also I'm trying to move away from the different branches I've created for different programs (docs/client/server/app) - it seems to be confusing. I'm going to be splitting things into various branches soon.

uncleashi, 2016-12-29:
Hi @schollz ,

Sure sounds good. I can make a separate repo for it which you can include into FIND documentation.
Also, I will include .apk file too as requested by someone so that developers can get started quickly.
Let me know if you need any help from my end to document it into FIND documentation? 
Will push the repo tonight. :)
Thanks

schollz, 2016-12-29:
@uncleashi, great! I'm excited to try it.

As for documentation, I'm happy to write it up, or you can submit a PR to the documentation page. The documentation pages are actually moved out of this repo now, and moved to [https://bitbucket.org/schollz/find-web](https://bitbucket.org/schollz/find-web/src/6004647945c532d97109eb201740291399fb61d9/docs/content?at=master) (I like to have stuff in both Bitbucket and Github since I don't like relying on just one service...).

uncleashi, 2016-12-30:
Hi @schollz ,

The initial commit of client is pushed now.. You can clone it and test. :)
More features coming soon.

schollz, 2016-12-30:
@uncleashi , I tested it out a little bit! Fantastic work, you really made a stunning app. 

I'll test it out a bit more before submitting to the Play store.

uncleashi, 2017-01-12:
Hi @schollz ,

Any feedback on it after your testing? Accordingly I'll add other features to it
Thanks

schollz, 2017-01-12:
@uncleashi 

Yeah, here are my thoughts so far:

1. Its *really* cool that you were able to do tracking when the the app is the background, and even when the phone is off! This is great! Along these lines, the app should probably tell the user it is running in the background, i.e. in the phone status bar.
2. There seems to be an issue with WiFi scanning. The app is submitting duplicate fingerprints for the same mac address - for example just see [the latest fingerprint from my test with the new app](https://ml.internalpositioning.com/lastfingerprint?group=rz&user=zzz), versus the [the same place with the old app](https://ml.internalpositioning.com/lastfingerprint?group=rzz&user=zack). You can see that 'c4:0a:cb:2c:27:1c' is a real MAC in both, but it seems like its overwriting all the other macs for some reason during the new app's scan.
3. There should be an obvious way to shut down the app so it stop running whatever it is doing (tracking or learning). Perhaps it can be done in the status bar when it is running in the background.
4. *Minor*: on startup a random group name should be assigned automatically so that multiple people downloading the app don't learn on the same group (so that everything works out of the box so to speak). (I can probably submit this myself as a PR so you don't need to focus on it).

uncleashi, 2017-01-13:
@schollz ,

Please find my response below:

#1. Yes, I have added that feature wherein if the scanning of any sort is going in the Background, then the user gets notified on Notif bar about it. This feature will be there in next push.

#2. Yeah, I too came across this minor bug last week during testing, it was a silly mistake on my end and even I found it when I was testing the app in a crowded environment. I have pushed the update code with this bug fix last week. Kindly take a pull and it'll be sorted for you. :) 

#3. Yeah, I can certainly add a stop scanning button into notif bar itself. Though I have to check once.

#4. Yeah, even one of my friend suggested me the same. :p Though I am thinking of what could be the unique name that we can give to the user on startup. One thought was, it could be their email ID which can be assigned on startup as GroupName and subsequently we can give a default username name too based on the email it itself. For ex:
I can ask for a sign in on app startup wherein the user can sign-in using its existing account. Let's say its "jhon@gmail.com".
We can assign that as the group name and assign username as "john".
The thought is to make it intuitive for the user even on the first run so that they don't have to look for what was the username or groupname. 

Let me know your thoughts on it or if you can think of any better way let me know.

uncleashi, 2017-01-15:
@schollz ,

Cool change the flow then as discussed.
Thanks

JammyDodger231, 2017-03-26:
@uncleashi, sorry to be a hassle but would it be possible to include an APK on your repo? I'm quite the Android Studio noob! I know you said about it earlier in the thread :) sorry for pestering

JammyDodger231, 2017-03-27:
Scratch my last comment, I managed to get up and running with just the repo and android studio, app seems to work great! :)

theRealMCQwerty, 2018-01-06:
@uncleashi 

I see your comment below about the notification in the notification bar but after having downloaded the latest version of your app, building and installing it I do not see anything in the notification bar. 

> Yes, I have added that feature wherein if the scanning of any sort is going in the Background, then the user gets notified on Notif bar about it. This feature will be there in next push.

I am very interested in your app, specifically for the background updates but on my moto x4 running android Oreo I get only very infrequent updates when the app is in the background. Maybe every 2 minutes but sometimes nothing for much longer periods of time.

- Is the code with for the notification bar in the version on github? If so, any idea why I don't see it.
- Any idea why the background updates are so unreliable, can I make it update the server regularly like every 5s even when in the background?

Thanks