schollz/find #125: Background process leaves room

kdschlosser, 2016-12-01:
When the application is put to the background it leaves the room. I noticed this when i was working on making the EventGhost Plugin. because the plugin shows an event happening when you leave and enter the room. and as soon as I put the app to the background an event comes up that I left the room.

schollz, 2016-12-03:
I'm not sure what you mean, that it leaves the room? The app should be classifying the locations, and the triggering that goes on with EventGhost should be the one involved with events. Can you send me a link to the EventGhost plugin?

kdschlosser, 2016-12-04:
sent it to ya in an e-mail. 

it only does it sometimes. but it will leave the location and then re enter it. 

but I am also told, now i have not tested for this issue as well. but it  falls into the same thing with the app being a background process.

if you go to you home screen. and then lock the device it stops updating.  but i am told that if you leave the app open then lock the device it works fine.

schollz, 2016-12-09:
Okay, this is probably a bug in the app. Can you give me your phone specs?

This is something that can be addressed in the new version: https://github.com/schollz/find/issues/124

kdschlosser, 2016-12-11:
OnePlus 2

uncleashi, 2016-12-12:
Yeah, I agree with Schollz, that it's a minor bug within the app. 
As far as I can think of the reason behind this behavior is that when you kill the app in the foreground, it basically kills all the resources and stop doing its task. Now once it goes in the background the service comes in and starts to scan for the WiFi APs. And since its restarting itself then, in that case, few readings, in the beginning, will have some error in terms of RSSI its receiving. And that is the reason why its showing to an EXIT event. I have come across this problem earlier when I was working on a WiFi module.
Ideally, once the app goes in background it should wait for few seconds till the adapter boots up and then start to streaming data to the server. Will try to fix this in upcoming app. 
Cheers

kdschlosser, 2016-12-12:
thanks. Just wanted to let ya know that i came across it when making that plugin for EventGhost. I can "smooth" it in the code for the plugin but I thought i should let ya know that i found it. wasn't sure if it was actually a bug or if it was just the way it has to work do to the changing to running in the background. but it does sound like it is something that can be sorted on your end. so i will not worry about implementing any smoothing. whichever would be easiest for you folks.

closed 2026-09-16 by kdschlosser