schollz/find #126: Not elevation aware

kdschlosser, 2016-12-01:
I have a 3 story house. and my office is right below my living room. when I am in my office it thinks i am in my living room.

schollz, 2016-12-03:
How many APs are in the vicinity?

The system *will* work with elevation. Basically, as long as the fingerprint is unique it will work. The fingerprint will be unique if you have enough routers in the vicinity. However, if there are not enough routers, or not enough learning, it will fail to resolve certain locations.

kdschlosser, 2016-12-04:
LOL @ "How many APs are in the vicinity?"

ummm.... 1

I live in the Rocky Mountains of Colorado. nearest neighbor is not that close. put it to ya this way. nearest gas station is 20 minutes away.

I do have several "spare" ap's (that means consumer level ones that like to drop connection under load)  I am guessing i should distribute them around the home only to generate a signal to use as reference.

schollz, 2016-12-04:
Yeah, its not going to work well for you then! 

You can just put your spare APs in different places, and that should help immensely. You just need them plugged in - you never need to connect to them. Do you have any Raspberry Pis? You can make those into APs pretty easily too.

kdschlosser, 2016-12-05:
Yeah after the e-mail i sent ya about how it functions i realized my error in posting this on the github... I can do one better then that

the esp8266's can act as an AP and will run on 2 coin sell batteries :-D

and they are only 3 bucks a piece

and I have about 10 of em sitting in a bad in my desk. LOL

I wonder if doing somehting like putting one in each room would do anything in terms of accuracy. because that would be one hell of a way to go about it. super low cost and if you didn't want to plug it in you could do that. tho i do not know how long an esp would run on a couple of coin cells. but a wall wart that will do 3.3v is not that expensive. just have to get a capacitor inline if a super cheap one is used (smooth out the AC bounce)

schollz, 2016-12-07:
Thats a great idea. I didn't know you can setup esp8266's as an AP! I'd appreciate any code/tutorial you have on this.

kdschlosser, 2016-12-07:
I have not done it in some time. and i set it into ap mode when i had an arduino connected to it and using the factory AT firmware...

it's actually quite cool the AT firmware. reminds me when i got my first modem 2400 baud w00 h00 then i got a 14400 that i ran at 19200 and the thing got so hot it melted the rubber feet on it. LOL

BBS days.

when i get around to messing with it i will type out specific detailed instructions

schollz, 2016-12-09:
That'd be great - I look forward to that.

closed 2016-12-09 by schollz