schollz/find #131: Get location history with /location

schollz, 2016-12-03:


closed 2016-12-03 by schollz

victorhooi, 2017-08-17:
Just out of curiosity - was there a reason this ticket was closed?

I checked the API docs - https://www.internalpositioning.com/api/#get-location - and it seems like GET /location doesn't currently return history - so I'm guessing you didn't end up doing this?

I would love to see history - is the current approach to just poll the server on a periodic basis (e.g. every 30 seconds) to pull current location of all devices? Or is there maybe a better way? (Some way of publishing it?).

schollz, 2017-08-18:
@victorhooi There is not yet a way of publishing the history but you can use `GET /location` to retrieve the history. Just use `/location?group=GROUP&n=100` to get the last 100 records for every user.

victorhooi, 2017-08-18:
Interesting - so we do store historical location records keyed by user?

Is there any way to get the state at a given time? (e.g. here were the X devices detected, and their locations at <timestamp>?)

Curious also on your thoughts of if BoltDB is the best place to store time-series  historical location data, or something more time-series oriented like InfluxDB would work for that part?

schollz, 2017-08-20:
Yes, historical locations are all stored. To get states at a given time you have to search through them - there is no way to do this in BoltDB except parsing all.

In retrospect I think something like InfluxDB would be great, but I don't have plans to do a rewrite for that.