schollz/find #137: Error- posterior.go: Not in network

uncleashi, 2017-01-07:
Hi @schollz ,

I am testing the FIND server on Azure VM for the FIND android client app development from past few days. It was working fine till yesterday and I even made some 30 mock locations by learning. But it stopped working this morning and throwing following errors in log. 
Also, I am curious to know did you check how many simultaneous cell phones the server can track. And if there is some limitation, can you point them out? That would an interesting thing to know and I think you should include that into feature documentation.
Thanks

[image]

schollz, 2017-01-07:
@uncleashi 

Not sure why your getting an error in the log - I'd have to see the raw fingerprints. It usually means that they aren't sending any fingerprints associated with the ones learned.

As for the questions:

1. I'm not sure - I've been able to track dozens with no problem. It gets slow to load all of them at once though.
2. This is the biggest limitation currently. Since the Naive Bayes implementation (which is the best classifier imo) relies on keeping a storage of  200 * N locations * M macs. The server needs more than 500MB of memory if you have ~20 locations with ~400 macs total. If you have the memory, then its not a worry. But there is room for optimization here.

Please let me know if this helps clear it up, I can certainly add it to the documentation!

uncleashi, 2017-01-07:
@schollz ,

A couple of things I didn't understand in your explanation:

1. "It gets slow to load all of them at once, though". Can you please explain this?

2. "200 * N locations * M macs". What is "200" here? So, lets if I have 1TB of storage available with me so in this case can I say that I don't have to worry about this issue?

Thanks

schollz, 2017-01-07:
@uncleashi 

Sure.

1. The slow part becomes calculating the current positions of everyone. To do this, the database iterates through every fingerprint, and adds the first fingerprint it finds (the latest) to a growing list of user fingerprints. Then it goes through and calculates the current position for each fingerprint of each user. Thus, if you open up a dashboard with 100 users, it will take a few seconds to iterate through the db and then take a few more seconds (or tens of seconds) to calculate everyone's current position. After that, though, the positions are cached. A speedup for this would be to use a real DB and not a keystore so that columns could be filtered more easily.

2. The issue is not storage, but RAM memory. The probability of seeing a mac address at a location is given in a 100-length array (to represent RSSI values -100 to -1).  There is an array for the probability of being in a location and seeing a mac, and there is a probability for being in a location and NOT seeing a mac, so there needs to be two of these 100-length arrays, i.e. 200. Every N locations needs to have a set of these arrays for all M macs, so thats why you have N x M x 200. I think you don't need to worry about this issue for most computers. I run on a DO droplet with only 200MB available memory, so I have to make sure people aren't running into this problem. If you have a modern computer (i.e. 4GB of Memory+), this is likely not an issue.

uncleashi, 2017-01-12:
Hi @schollz ,

That answer did certainly helped me in terms of understanding how the server works. 
Also, I am planning to do some stress testing on a standard server(2 cores, 4 GB RAM), to check:
a. How many locations can a standard server support.
b. How many simultaneous data stream it can handle

I think that will reveal a lot many unexplored things for us to work on and improve. :)
But I still did not understand the reason for that error and all of a sudden everything stopped working. 
Let me know your thoughts..
Thanks

schollz, 2017-01-12:
Sure, the current FIND server at ml.internalpositioning.com generally has ~120 unique users per day. Most users only have 3 to 20 rooms and 1 to 4 tracking users. Its been handling ~2 requests / second for the past 6 months. It uses about 300 MB of RAM (and a little bit of swap) and the CPU is usually < 10%.

The screenshot you have shows "Not in network" warning - is that what you are referring to as the error? This warning comes from the first step in the classification - first the server checks each mac address against the list of mac addresses it knows for each network (a network is just a connected network of all mac addresses that have co-appeared in a fingerprint). If *none* of the mac addresses in the current fingerprint match any in any network, then its clear to the server that it cannot be classified because there is no information on it. Then it will return a warning "Not in network". In the instance that you have a fingerprint with 10 mac addresses that the server has not seen, and 1 that the server has, then the server will still classify (just ignoring the 10 it hasn't seen before). Please let me know if I can explain further!

uncleashi, 2017-01-15:
@schollz ,

Did lil more research on the above error and this is what I found in logs. Have a look and let me know what exactly is it?
 
[image]

uncleashi, 2017-01-22:
Hi, 

Did you look into it, any luck?

schollz, 2017-01-22:
Can you send me your DB file?

uncleashi, 2017-01-25:
@schollz ,
 The .db format is not supported in github. :(

schollz, 2017-01-26:
You can just email it to me, zack@hypercubeplatforms.com. I'll take a look and see if I can replicate your error.

uncleashi, 2017-01-30:
Hi @schollz ,

mailed you the db last week. Did you go through it?

uncleashi, 2017-02-01:
Hi @schollz ,

I tried to open the .db file in /data folder but its throwing error. Could you please suggest a tool to view the db file ?
Thanks

schollz, 2017-02-02:
Hi @uncleashi 
Looks like its working for me, but I had to "Calculate" the parameters. Maybe you could try the same? Just goto the dashboard and click "Update" and try tracking again.

uncleashi, 2017-02-03:
@schollz ,

Yeah, its working now. Though I'll be doing some more testing to see if its behaving the same as it was earlier.
On the side note we should ideally run a service(say every 10min) which "Re-Calculate" the model inorder to avoid this error if at all its the root cause of the error.
Your views?

hdukh, 2017-10-28:
@schollz  @uncleashi 
Hi all, I have encountered the same problem, after investigation, I think it should be the Android app problem : find-client\app\src\main\java\com\find\wifitool\wifi\WifiIntentReceiver.java Line 62.
`JSONObject wifiResults = new JSONObject();`
When inserted wifiResults into the wifiResultsArray will lead to data coverage.