schollz/find #141: [Enhancement/RFC] Presence detection without a phone

arsaboo, 2017-01-22:
I understand FIND uses an Android device to measure the wifi signal strength for internal location mapping. Recently, I came across two services [Aerial](http://www.aerial.ai/) and [Aura](https://www.aurahome.com/) that sound very similar to FIND, but don't rely on ANY device for presence. Both are in beta and use wifi signals across the house to detect presence. It sounds like both of them rely on signal processing, something that FIND should be able to do. Wondering if it is possible to do something similar with FIND, so that we are not stuck with only Android devices. 

I would love to hear your thoughts on this.

schollz, 2017-01-22:
Hi @arsaboo, thanks for those links! I haven't really seen those specific vendors but I know people are making devices to do that.  They are pretty cool! Though, I have no idea how they are able to "detect ripples in wireless signals" to position people. It sounds like sonar, but in my experience its hard to do that with radio waves because of interference from all sorts of things (microwaves, tvs, etc.).

I've been working on something similar though: https://github.com/schollz/find-lf. In this case, however, it works by tracking WiFi probe requests sent out by wifi devices (like phones) but it doesn't require any special software on the phone. Also it only requires a few raspberry pis so it costs <$100.

arsaboo, 2017-01-22:
I did some digging around and found some links that will provide more insights. See http://people.csail.mit.edu/fadel/wivi/ for details on how to use wifi to detect human activity and more. Here is another related project that may be of interest to you that uses sonar http://rfcapture.csail.mit.edu/