schollz/find #142: invalid memory address or nil pointer dereference

karona75, 2017-01-25:
I've got the FIND API running locally (i.e on my laptop) and I am trying to track my self around the house.

The android app successfully posts the fingerprints for training, however, on enabling tracking I get this:

```

2017/01/25 15:23:32 [Recovery] panic recovered:
POST /track HTTP/1.1
Host: 192.168.1.17:8003
Accept: application/json, text/javascript, */*; q=0.01
Accept-Encoding: gzip, deflate
Accept-Language: en-GB,en-US;q=0.8
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Origin: file://
User-Agent: Mozilla/5.0 (Linux; Android 6.0.1; SM-N9005 Build/MOB30Z; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/51.0.2704.106 Mobile Safari/537.36
X-Requested-With: com.hcp.find

runtime error: invalid memory address or nil pointer dereference
/usr/lib/go-1.6/src/runtime/panic.go:443 (0x491909)
        gopanic: reflectcall(nil, unsafe.Pointer(d.fn), deferArgs(d), uint32(d.siz), uint32(d.siz))
/usr/lib/go-1.6/src/runtime/panic.go:62 (0x48ffc5)
        panicmem: panic(memoryError)
/usr/lib/go-1.6/src/runtime/sigpanic_unix.go:24 (0x4a63aa)
        sigpanic: panicmem()
/usr/lib/go-1.6/src/fmt/print.go:189 (0x4e0acc)
        Fprintf: n, err = w.Write(p.buf)
/home/lemonsnowcone/Source/find/rf.go:85 (0x44a2db)
        rfClassify: fmt.Fprintf(conn, group+"="+tempFile)
/home/lemonsnowcone/Source/find/fingerprint.go:267 (0x41519c)
        trackFingerprint: userJSON.Rf = rfClassify(strings.ToLower(jsonFingerprint.Group), jsonFingerprint)
/home/lemonsnowcone/Source/find/fingerprint.go:138 (0x412b2a)
        trackFingerprintPOST: message, success, locationGuess, bayes, svm, rf := trackFingerprint(jsonFingerprint)
/home/lemonsnowcone/Source/find/go/src/github.com/gin-gonic/gin/context.go:97 (0x5bfd5a)
        (*Context).Next: c.handlers[c.index](c)
/home/lemonsnowcone/Source/find/go/src/github.com/gin-gonic/contrib/sessions/sessions.go:65 (0x683a73)
        Sessions.func1: c.Next()
/home/lemonsnowcone/Source/find/go/src/github.com/gin-gonic/gin/context.go:97 (0x5bfd5a)
        (*Context).Next: c.handlers[c.index](c)
/home/lemonsnowcone/Source/find/go/src/github.com/gin-gonic/gin/recovery.go:45 (0x5d2991)
        RecoveryWithWriter.func1: c.Next()
/home/lemonsnowcone/Source/find/go/src/github.com/gin-gonic/gin/context.go:97 (0x5bfd5a)
        (*Context).Next: c.handlers[c.index](c)
/home/lemonsnowcone/Source/find/go/src/github.com/gin-gonic/gin/logger.go:72 (0x5d1c02)
        LoggerWithWriter.func1: c.Next()
/home/lemonsnowcone/Source/find/go/src/github.com/gin-gonic/gin/context.go:97 (0x5bfd5a)
        (*Context).Next: c.handlers[c.index](c)
/home/lemonsnowcone/Source/find/go/src/github.com/gin-gonic/gin/gin.go:284 (0x5c6e92)
        (*Engine).handleHTTPRequest: context.Next()
/home/lemonsnowcone/Source/find/go/src/github.com/gin-gonic/gin/gin.go:265 (0x5c6ac7)
        (*Engine).ServeHTTP: engine.handleHTTPRequest(c)
/usr/lib/go-1.6/src/net/http/server.go:2081 (0x54323e)
        serverHandler.ServeHTTP: handler.ServeHTTP(rw, req)
/usr/lib/go-1.6/src/net/http/server.go:1472 (0x53faee)
        (*conn).serve: serverHandler{c.server}.ServeHTTP(w, w.req)
/usr/lib/go-1.6/src/runtime/asm_amd64.s:1998 (0x4c2151)
        goexit: BYTE    $0x90   // NOP

```

The above error is from running the random forest classification method. I had a feeling this may have been cause by my own faulty configs. However, after rolling out a clean build the problem persists.

schollz, 2017-01-26:
Do you know if the RF server is correctly processing fingerprints? If you run it in another terminal you should be able to track its output and whether its spitting out errors.

anilet, 2017-01-28:
Hello,
I am getting a similar error after running the server for several hours.
2017/01/28 20:06:50 [Recovery] panic recovered:
POST /track HTTP/1.1
Host: justtv.duckdns.org:18003
Accept-Encoding: gzip
Connection: Keep-Alive
Content-Length: 311
Content-Type: application/json; charset=UTF-8
User-Agent: okhttp/3.2.0

page 154 already freed
/usr/local/go/src/runtime/panic.go:458 (0x47b4c3)
        gopanic: reflectcall(nil, unsafe.Pointer(d.fn), deferArgs(d), uint32(d.siz), uint32(d.siz))
/home/anilroy/src/github.com/boltdb/bolt/freelist.go:121 (0x57e990)
        (*freelist).free: panic(fmt.Sprintf("page %d already freed", id))
/home/anilroy/src/github.com/boltdb/bolt/node.go:363 (0x58182c)
        (*node).spill: tx.db.freelist.free(tx.meta.txid, tx.page(node.pgid))
/home/anilroy/src/github.com/boltdb/bolt/bucket.go:570 (0x576f4b)
        (*Bucket).spill: if err := b.rootNode.spill(); err != nil {
/home/anilroy/src/github.com/boltdb/bolt/tx.go:163 (0x584985)
        (*Tx).Commit: if err := tx.root.spill(); err != nil {
/home/anilroy/src/github.com/boltdb/bolt/db.go:605 (0x57c42d)
        (*DB).Update: return t.Commit()
/home/anilroy/tmp/find/fingerprint.go:130 (0x4104ac)
        putFingerprintIntoDatabase: })
/home/anilroy/tmp/find/fingerprint.go:227 (0x411ecb)
        trackFingerprint: putFingerprintIntoDatabase(fullFingerprint, "fingerprints-track")
/home/anilroy/tmp/find/fingerprint.go:138 (0x41088a)
        trackFingerprintPOST: message, success, locationGuess, bayes, svm, rf := trackFingerprint(jsonFingerprint)
/home/anilroy/src/github.com/gin-gonic/gin/context.go:97 (0x58a22a)
        (*Context).Next: c.handlers[c.index](c)
/home/anilroy/src/github.com/gin-gonic/contrib/sessions/sessions.go:65 (0x62d159)
        Sessions.func1: c.Next()
/home/anilroy/src/github.com/gin-gonic/gin/context.go:97 (0x58a22a)
        (*Context).Next: c.handlers[c.index](c)
/home/anilroy/src/github.com/gin-gonic/gin/recovery.go:45 (0x59a00a)
        RecoveryWithWriter.func1: c.Next()
/home/anilroy/src/github.com/gin-gonic/gin/context.go:97 (0x58a22a)
        (*Context).Next: c.handlers[c.index](c)
/home/anilroy/src/github.com/gin-gonic/gin/logger.go:72 (0x5990ff)
        LoggerWithWriter.func1: c.Next()
/home/anilroy/src/github.com/gin-gonic/gin/context.go:97 (0x58a22a)
        (*Context).Next: c.handlers[c.index](c)
/home/anilroy/src/github.com/gin-gonic/gin/gin.go:284 (0x5905be)
        (*Engine).handleHTTPRequest: context.Next()
/home/anilroy/src/github.com/gin-gonic/gin/gin.go:265 (0x58fea0)
        (*Engine).ServeHTTP: engine.handleHTTPRequest(c)
/usr/local/go/src/net/http/server.go:2202 (0x51ad8d)
        serverHandler.ServeHTTP: handler.ServeHTTP(rw, req)
/usr/local/go/src/net/http/server.go:1579 (0x5176f7)
        (*conn).serve: serverHandler{c.server}.ServeHTTP(w, w.req)
/usr/local/go/src/runtime/asm_amd64.s:2086 (0x4aabf1)
        goexit: BYTE    $0x90   // NOP

running the server with or without -rf  flag makes no difference. I do have libsvm installed.
@karona75 how did you solve the issue?

schollz, 2017-01-28:
Thanks.

Couple of questions @karona75 and @anilet : 

What version of FIND server or are you running? Did you build it yourself, and if so when?

I've been seeing this issue too. It is new to me, but I think it is related to this: https://github.com/boltdb/bolt/issues/348. There was a pretty recent fixes for Bolt, that may help. Or I could vendor an old version. I probably should have used vendoring from the beginning because pulling the master makes it really hard to figure out what version is messing you guys up.

anilet, 2017-01-28:
I built it myself today, the server says "version  build devdevde" when run.
One thing I noticed during install is that there is no findserver binary to run only find ?

schollz, 2017-01-28:
The binary name will depend on how its built - go build will build it with the name of the folder. make will build it as findserver.

And what version of Go are you using?

anilet, 2017-01-28:
go version go1.7.3 linux/amd64 on Ubuntu 16.04

schollz, 2017-01-28:
Just in case, try this:

```
go get -u github.com/boltdb/bolt
```

and then rebuild FIND.

Also, how many times have you encountered this error?

anilet, 2017-01-28:
It was running ok for about 2 hours and both of my phones were tracked well, then suddenly I started getting these errors and I could not even start the server anymore without deleting the database and other files in data folder.

schollz, 2017-01-28:
Okay, I would just try again and let me know if you get the same error. If so, I think vendoring BoltDB will be required.

anilet, 2017-01-28:
same error again
On the front end locations are loaded but no users

2017/01/28 21:42:59 [Recovery] panic recovered:
POST /track HTTP/1.1
Host: justtv.duckdns.org:18003
Accept-Encoding: gzip
Connection: Keep-Alive
Content-Length: 389
Content-Type: application/json; charset=UTF-8
User-Agent: okhttp/3.2.0

page 154 already freed
/usr/local/go/src/runtime/panic.go:458 (0x47b4c3)
        gopanic: reflectcall(nil, unsafe.Pointer(d.fn), deferArgs(d), uint32(d.siz), uint32(d.siz))
/home/anilroy/tmp/src/github.com/boltdb/bolt/freelist.go:121 (0x57e990)
        (*freelist).free: panic(fmt.Sprintf("page %d already freed", id))
/home/anilroy/tmp/src/github.com/boltdb/bolt/node.go:363 (0x58182c)
        (*node).spill: tx.db.freelist.free(tx.meta.txid, tx.page(node.pgid))
/home/anilroy/tmp/src/github.com/boltdb/bolt/bucket.go:570 (0x576f4b)
        (*Bucket).spill: if err := b.rootNode.spill(); err != nil {
/home/anilroy/tmp/src/github.com/boltdb/bolt/tx.go:163 (0x584985)
        (*Tx).Commit: if err := tx.root.spill(); err != nil {
/home/anilroy/tmp/src/github.com/boltdb/bolt/db.go:605 (0x57c42d)
        (*DB).Update: return t.Commit()
/home/anilroy/tmp/find/fingerprint.go:130 (0x4104ac)
        putFingerprintIntoDatabase: })
/home/anilroy/tmp/find/fingerprint.go:227 (0x411ecb)
        trackFingerprint: putFingerprintIntoDatabase(fullFingerprint, "fingerprints-track")
/home/anilroy/tmp/find/fingerprint.go:138 (0x41088a)
        trackFingerprintPOST: message, success, locationGuess, bayes, svm, rf := trackFingerprint(jsonFingerprint)
/home/anilroy/tmp/src/github.com/gin-gonic/gin/context.go:97 (0x58a22a)
        (*Context).Next: c.handlers[c.index](c)
/home/anilroy/tmp/src/github.com/gin-gonic/contrib/sessions/sessions.go:65 (0x62d159)
        Sessions.func1: c.Next()
/home/anilroy/tmp/src/github.com/gin-gonic/gin/context.go:97 (0x58a22a)
        (*Context).Next: c.handlers[c.index](c)
/home/anilroy/tmp/src/github.com/gin-gonic/gin/recovery.go:45 (0x59a00a)
        RecoveryWithWriter.func1: c.Next()
/home/anilroy/tmp/src/github.com/gin-gonic/gin/context.go:97 (0x58a22a)
        (*Context).Next: c.handlers[c.index](c)
/home/anilroy/tmp/src/github.com/gin-gonic/gin/logger.go:72 (0x5990ff)
        LoggerWithWriter.func1: c.Next()
/home/anilroy/tmp/src/github.com/gin-gonic/gin/context.go:97 (0x58a22a)
        (*Context).Next: c.handlers[c.index](c)
/home/anilroy/tmp/src/github.com/gin-gonic/gin/gin.go:284 (0x5905be)
        (*Engine).handleHTTPRequest: context.Next()
/home/anilroy/tmp/src/github.com/gin-gonic/gin/gin.go:265 (0x58fea0)
        (*Engine).ServeHTTP: engine.handleHTTPRequest(c)
/usr/local/go/src/net/http/server.go:2202 (0x51ad8d)
        serverHandler.ServeHTTP: handler.ServeHTTP(rw, req)
/usr/local/go/src/net/http/server.go:1579 (0x5176f7)
        (*conn).serve: serverHandler{c.server}.ServeHTTP(w, w.req)
/usr/local/go/src/runtime/asm_amd64.s:2086 (0x4aabf1)
        goexit: BYTE    $0x90   // NOP

schollz, 2017-01-28:
Is this when you try from the very beginning on a new database?

anilet, 2017-01-28:
No I copied the old database.
Now that I have deleted the database and everything is working, will add few more locations to see how it goes

anilet, 2017-01-29:
Seems like the server is running ok with new database. Its been running for 24 hours

schollz, 2017-01-31:
Okay, I've found that this is a bolt issue. You should rebuild after you replace the bolt library with https://github.com/boltdb/bolt/releases/tag/v1.2.1.

karona75, 2017-02-06:
This seems to have solved the problem. Replaced bolt with v1.2.1

closed 2017-02-06 by karona75

schollz, 2017-02-06:
@karona75 Sorry I forgot to update this - I think I pinpointed the problem in bolt and it was reverted in the master of bolt itself (see https://github.com/boltdb/bolt/pull/654). You should be able to use the bolt master now, you can update it with `go get -u -v github.com/boltdb/bolt`.