schollz/find #143: Running FIND as a background service

karona75, 2017-01-25:
I've moved the app to be managed by supervisord and I run into the following error when trying to track a phone/client in a trained location:

```
-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----
   _________  _____
  / __/  _/ |/ / _ \  ______ _____  _____ ____
 / _/_/ //    / // / (_-< -_) __/ |/ / -_) __/
/_/ /___/_/|_/____/ /___|__/_/  |___/\__/_/

(version  build devdevde) is up and running on http://<digital ocean IP>:8003
-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----+-----

[31m2017/01/25 15:52:50 [Recovery] panic recovered:
POST /track HTTP/1.1
Host: <digital ocean IP>:8003
Accept: application/json, text/javascript, */*; q=0.01
Accept-Encoding: gzip, deflate
Accept-Language: en-GB,en-US;q=0.8
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Origin: file://
User-Agent: Mozilla/5.0 (Linux; Android 6.0.1; SM-N9005 Build/MOB30Z; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/51.0.2704.106 Mobile Safari/537.36
X-Requested-With: com.hcp.find

invalid page type: 24: 10
/usr/lib/go-1.6/src/runtime/panic.go:443 (0x491909)
	gopanic: reflectcall(nil, unsafe.Pointer(d.fn), deferArgs(d), uint32(d.siz), uint32(d.siz))
/home/demo/find/go/src/github.com/boltdb/bolt/cursor.go:256 (0x5abd55)
	(*Cursor).search: panic(fmt.Sprintf("invalid page type: %d: %x", p.id, p.flags))
/home/demo/find/go/src/github.com/boltdb/bolt/cursor.go:159 (0x5ab27a)
	(*Cursor).seek: c.search(seek, c.bucket.root)
/home/demo/find/go/src/github.com/boltdb/bolt/bucket.go:267 (0x5a7b54)
	(*Bucket).Get: k, v, flags := b.Cursor().seek(key)
/home/demo/find/parameters.go:152 (0x45f2ca)
	openParameters.func1: v := b.Get([]byte("fullParameters"))
/home/demo/find/go/src/github.com/boltdb/bolt/db.go:629 (0x5af2e9)
	(*DB).View: err = fn(t)
/home/demo/find/parameters.go:155 (0x41e26d)
	openParameters: })
/home/demo/find/posterior.go:14 (0x436bc9)
	calculatePosterior: ps, _ = openParameters(res.Group)
/home/demo/find/fingerprint.go:213 (0x4148b0)
	trackFingerprint: locationGuess1, bayes := calculatePosterior(jsonFingerprint, *NewFullParameters())
/home/demo/find/fingerprint.go:138 (0x412b2a)
	trackFingerprintPOST: message, success, locationGuess, bayes, svm, rf := trackFingerprint(jsonFingerprint)
/home/demo/find/go/src/github.com/gin-gonic/gin/context.go:97 (0x5bfd5a)
	(*Context).Next: c.handlers[c.index](c)
/home/demo/find/go/src/github.com/gin-gonic/contrib/sessions/sessions.go:65 (0x683a73)
	Sessions.func1: c.Next()
/home/demo/find/go/src/github.com/gin-gonic/gin/context.go:97 (0x5bfd5a)
	(*Context).Next: c.handlers[c.index](c)
/home/demo/find/go/src/github.com/gin-gonic/gin/recovery.go:45 (0x5d2991)
	RecoveryWithWriter.func1: c.Next()
/home/demo/find/go/src/github.com/gin-gonic/gin/context.go:97 (0x5bfd5a)
	(*Context).Next: c.handlers[c.index](c)
/home/demo/find/go/src/github.com/gin-gonic/gin/logger.go:72 (0x5d1c02)
	LoggerWithWriter.func1: c.Next()
/home/demo/find/go/src/github.com/gin-gonic/gin/context.go:97 (0x5bfd5a)
	(*Context).Next: c.handlers[c.index](c)
/home/demo/find/go/src/github.com/gin-gonic/gin/gin.go:284 (0x5c6e92)
	(*Engine).handleHTTPRequest: context.Next()
/home/demo/find/go/src/github.com/gin-gonic/gin/gin.go:265 (0x5c6ac7)
	(*Engine).ServeHTTP: engine.handleHTTPRequest(c)
/usr/lib/go-1.6/src/net/http/server.go:2081 (0x54323e)
	serverHandler.ServeHTTP: handler.ServeHTTP(rw, req)
/usr/lib/go-1.6/src/net/http/server.go:1472 (0x53faee)
	(*conn).serve: serverHandler{c.server}.ServeHTTP(w, w.req)
/usr/lib/go-1.6/src/runtime/asm_amd64.s:1998 (0x4c2151)
	goexit: BYTE	$0x90	// NOP
```

Training works fine, however tracking does not. Running this as a manual call in an active ssh session does not result in this issue. Running the app via nohup or systemd also results in the same issue. 

I'm a little hesitant to go down the docker route as I have no experience with container services.

schollz, 2017-01-26:
You shouldn't need Docker. I'm running the program on DO just fine.

Is this a brand new DB? Or did you transfer it from another FIND instance?

schollz, 2017-01-31:
Okay, I've found that this is a bolt issue. You should rebuild after you replace the bolt library with https://github.com/boltdb/bolt/releases/tag/v1.2.1.

karona75, 2017-02-06:
Rolling back bolt solve the problem.

closed 2017-10-03 by schollz