schollz/find #144: Server stops publishing MQTT messages after around an hour

lwis, 2017-01-28:
I'm using a custom MQTT broker with the docker image, and I don't see any errors in the logs - it just stops publishing to the broker after some time, only fixed by a restart. Logs show that it's still active and processing fingerprints.

schollz, 2017-01-28:
That is strange. I wonder if it is related to https://github.com/schollz/find/issues/142, but I don't know why you don't see a panic in the log.

tavalin, 2017-01-28:
I had something similar when I had network issues. I was wondering if the mqtt client tried to reconnect to the broker if the connection dropped.

lwis, 2017-01-28:
@schollz I just double checked, I'm not seeing anything other than `POST    /track`, and `fingerprint.go:229` debug logs.

I can deploy changes/try anything to help debug.

@tavalin I think it's unlikely as my mqtt broker is hosted on the same box as FIND.

lwis, 2017-01-28:
@schollz update;

I see `Socket error on client RylwF, disconnecting.` from mosquitto when the connection stops - I've also noticed that FIND subscribes to # - why is this?

schollz, 2017-01-31:
FIND subscribes to # so any published fingerprint can be processed - see https://www.internalpositioning.com/server/#mqtt-support. Since MQTT is non-anonymous, it requires people to have passwords to subscribe/post. Instead of subscribing to each individual channel for each user/group, FIND just uses admin privileges to subscribe to all of them at once and delineate which ones its listening to when the fingerprint comes in.

I'm still puzzled by this error. Are you using a different MQTT broker than mosquito?

lwis, 2017-01-31:
@schollz I'm using mosquitto, and FIND is subscribing to # on a busy broker, so it's receiving a high volume of messages it isn't interested in.

tavalin, 2017-01-31:
Instead of subscribing to everything, Find could subscribe to the following topics:

```
+/location/+
+/track/+
+/learn/+
```

lwis, 2017-01-31:
@tavalin that would improve things dramatically

lwis, 2017-02-03:
Closing this for now as clearing my data appeared to fix it, not sure what could have caused it, though.

lwis, 2017-02-04:
Actually, maybe just a red herring :(

schollz, 2017-02-04:
I like @tavalin's fix - I will try to implement that when I get a chance.

@lwis do you see any logs about the issue? Could you try the non-docker version?

lwis, 2017-02-04:
@schollz nothing abnormal in the logs, just debug lines from `fingerprint.go`.

Can do.

closed 2018-02-12 by lwis