schollz/find #145: find on LG hom bot

cribskip, 2017-02-04:
Hi,

as many LG hom bots (automatic robot cleaners) are capable of using Wifi using a USB dongle, I gave it a try and has been successful but not to the fullest.

Find starts fine, searching works using the '--iwlist'-option but submitting to the demo broker fails because of missing x509 certificates:

> most2120# cd /usr/rbin/
most2120# ./findclient -e -g group -u user -s https://ml.internalpositioning.com --iwlist               
Enter group: 
Instead of typing next time, add '-g '
Enter location: 
Instead of typing next time, add '-l '
DEPRECATED Action signature.  Must be `cli.ActionFunc`.  This is an error in the application.  Please contact the distributor of this application if this is not you.  See https://github.com/urfave/cli/blob/master/CHANGELOG.md#deprecated-cli-app-action-signature
16:46:40.627 main - NOTI 001 You can see fewer messages by adding --nodebug
16:46:40.639 main - NOTI 002 User: user
16:46:40.651 main - NOTI 003 Group: 
16:46:40.660 main - NOTI 004 Server: https://ml.internalpositioning.com/learn
16:46:40.671 main - NOTI 005 Location: 
16:46:40.678 main - NOTI 006 Running 3 times (you can run more using '-c SOMENUM'). Please wait...
16:46:40.684 main - INFO 007 Scanning Wifi
16:46:40.698 scanWifi - INFO 008 Gathering fingerprint with '/sbin/iwlist wlan0 scan'
16:46:41.938 main - INFO 009 Processing 31 lines out output
16:46:41.945 main - INFO 00a Sending fingerprint to https://ml.internalpositioning.com/learn
**16:46:42.304 main - CRIT 00b Post https://ml.internalpositioning.com/learn: x509: failed to load system roots and no roots provided**

Would be easy to place the certs inside /etc/ssl - but this filesystem is read only :-(

Is there a option for the path to change to e.g. /usr/etc/ssl?

Best regards,
Sascha

schollz, 2017-02-04:
I'm not sure about how to modify `/etc/ssl` to fix that error.

A quick fix is to use the non-HTTPS public server, with the flag `--server http://ml2.internalpositioning.com`. Another possibility is that you could host the server yourself at home (since I assume your robot cleaner is not leaving the house). Then you don't need to worry about http security at all.

cribskip, 2017-02-06:
Thanks for the hint. Long term, I'm planning on deploying my own server, thanks :)

Nonetheless, now it shows a new error with the new server:

> most2120# ./findclient --server http://ml2.internalpositioning.com --iwlist
Enter group: LG
Instead of typing next time, add '-g lg'
DEPRECATED Action signature.  Must be `cli.ActionFunc`.  This is an error in the application.  Please contact the distributor of this application if this is not you.  See https://github.com/urfave/cli/blob/master/CHANGELOG.md#deprecated-cli-app-action-signature
09:06:36.132 main - NOTI 001 You can see fewer messages by adding --nodebug
09:06:36.133 main - NOTI 002 User: user
09:06:36.134 main - NOTI 003 Group: lg
09:06:36.135 main - NOTI 004 Server: http://ml2.internalpositioning.com/track
09:06:36.136 main - NOTI 005 Running 3 times (you can run more using '-c SOMENUM'). Please wait...
09:06:36.137 main - INFO 006 Scanning Wifi
09:06:36.138 scanWifi - INFO 007 Gathering fingerprint with '/sbin/iwlist wlan0 scan'
09:06:37.397 main - INFO 008 Processing 44 lines out output
09:06:37.415 main - INFO 009 Sending fingerprint to http://ml2.internalpositioning.com/track
09:06:37.760 main - CRIT 00a Something wrong with server

Best regards,
Sascha

schollz, 2017-02-06:
Yeah, it looks like you need to do some learning first, since that group doesn't exist yet. Sorry the error message is opaque

closed 2022-07-03 by cribskip