schollz/find #146: MAC filtering is not working

kamk123, 2017-02-13:
Hi,

we are trying to locate moving vehicle in the indoor area using RPi. To achieve this we are using MAC filtering feature as stated on the FIND official website. when we provide the option of 'macs.json' to the server executable, server unable to locate the device & provide empty string instead of proper location.

could you please help us to achieve MAC filtering properly?

Thanks for your help!

schollz, 2017-02-13:
Can you send the content of your `macs.json` and the error message from the server?

kamk123, 2017-02-14:
Hi,

Thank you for prompt response!

Contents of 'macs.json' are as follows:

{
	"c4:6e:1f:50:f0:22":true,
	"00:1e:a6:09:fa:21":true,
	"c4:6e:1f:d4:db:e8":true,
	"64:66:b3:53:cf:48":true,
	"e8:94:f6:2a:62:6a":true,
	"00:24:b2:94:ba:d6":true,
	"00:1f:33:8c:5d:fa":true
}

we are using 'mqtt' client to get the locations published by FIND server. when we provide 'macs.json' to server  we didn't get location.
we are using following command to run the FIND server with MAC filtering option:
./findserver -mqtt 127.0.0.1:1883 -mqttadmin ADMIN -mqttadminpass ADMIN_PASS -mosquitto `pgrep mosquitto` -filter macs.json

 if we run the server without macs.json, 'mqtt' client can able to receive locations correctly.

Thanks.

schollz, 2017-02-14:
That looks correct. Would you be able to send me your database file as well? You can just email it to me at hypercube.platforms@gmail.com.