schollz/find #147: REST API to get all configured locations?

kiwiandroiddev, 2017-02-13:
Hi,

I wondered if there could be a new API for retrieving the list of all configured locations for a group on a server?

Something like `GET /locations?group=my_group` that returns:
```json
{
   "message":"Correctly found locations.",
   "success":true,
   "locations":["bed bath", "bedroom", "car", "dining"]
}
```

It seems that this data can be obtained by appropriating the existing `/location` call for some known user, but not without getting a lot of extra irrelevant data too (signal strengths) which isn't ideal for mobile clients.

schollz, 2017-02-14:
Your right, I avoided this since I figured folks would do as you suggest. I can certainly add, it's easy. Ill put it up later this week!

TheMegaTB, 2017-02-15:
It would also be great if the `location` array could contain some metadata like the amount of stored fingerprints for each location.

schollz, 2017-02-16:
@TheMegaTB @kiwiandroiddev 

I've added it now, see if this will work for you: https://www.internalpositioning.com/api/#get-locations

TheMegaTB, 2017-02-16:
Just out of curiosity, how is the accuracy calculated?
(I want to display some kind of progress bar for learning and am searching for the best way to do it)

TheMegaTB, 2017-02-16:
@schollz thanks for the quick implementation of that feature! One more thing that prevents it from being used though is the missing Cross-Origin resource sharing header. Please have a look at #148 for the details.

schollz, 2017-02-16:
@TheMegaTB The accuracy is calculated for the Naive Bayes implementation - it uses cross validation against a sequestered test set against the set used for learning. Its useful for figuring out which places aren't learned well.

TheMegaTB, 2017-02-16:
It should be noted that newly learned locations are not returned by `/locations` right after learning and it requires a call to the `/calculate` endpoint.

schollz, 2017-02-16:
Thanks, I updated the docs with that note