schollz/find #148: CORS prevents access to API from JavaScript (not set for POST requests)

TheMegaTB, 2017-02-16:
It is not possible to run HTTP requests to the FIND server due to the Cross-Origin resource sharing settings (or rather non-set settings). When I load the page directly by accessing `localhost:18003/locations` using a `GET` request the CORS header is properly set. But when sending a `POST` request from JS (XMLHttpRequest) the header properties are missing, effectively preventing the request from being processed. Using the same JS but changing the request type to `GET` works.
This is the request and response resulting from a JS POST request:
[image]

schollz, 2017-02-16:
Thanks, never ran into this before! I added it properly to the routes: https://github.com/schollz/find/commit/0f43ca86d2f628366d39f505c768240050487cfd

Try it and let me know if that works for you!

TheMegaTB, 2017-02-16:
Doesn't look like it works. I ran the Docker container and just to make sure also compiled it from sources but still the same response header...
To reproduce this issue run the following in the GChrome dev console:
```
function httpAsync(theUrl, callback, type, payload) {
    var xmlHttp = new XMLHttpRequest();
    xmlHttp.onreadystatechange = function() {
        if (xmlHttp.readyState == 4 && xmlHttp.status == 200)
            callback(xmlHttp.responseText);
    }
    xmlHttp.open(type ? type : "GET", theUrl, true); // true for asynchronous
    xmlHttp.send(payload);
}
httpAsync("http://localhost:8003/locations", function (res) {console.log(res)}, "POST", "{'group': 'SomeGroup'}");
```

schollz, 2017-02-16:
Ah, its because `/locations` is a `GET` request, not a `POST`.

TheMegaTB, 2017-02-16:
Interestingly enough that command returns me the dashboard landing page instead of the expected locations result, when ran from the same origin (aka `localhost:8003/dashboard/SomeGroup`)

TheMegaTB, 2017-02-16:
Well then this is fixed. Thanks for the quick fix there.

closed 2017-02-16 by TheMegaTB