schollz/find #149: Sending data to /learn without "time" property overwrites previous data

TheMegaTB, 2017-02-16:
I am inserting data by using the `/learn` endpoint of the API and for simplicity reason's I didn't send the time because I read somewhere that it isn't required. Turns out though that apparently it is indeed required since otherwise the server simply overwrites previously send data (`/locations` and the dashboard return only one fingerprint that has been learned so that's my assumption there). 

Since I am learning from within the browser using the API of my IoT device it doesn't matter all that much but it would be very convenient if the `time` property could be omitted, since it might be rather hard to get your hands on the current time on a IoT device.

schollz, 2017-02-16:
You don't need to send time. If you don't send time it will use the current time when it inserts into the database: https://github.com/schollz/find/blob/master/fingerprint.go#L122-L124. 

And, since it stores fingerprints in a map by the timestamp, it will only overwrite if the timestamp is exactly the same.

That said, I'm not sure why you are experiencing the overwriting? How do you know its overwriting? Are you sending fingerprints with the same timestamps? Are you sending them all at the same time? Its possible, I suppose, that if you send tons of fingerprints at the same time (on the order of sub-nanoseconds) they will overwrite each other.

TheMegaTB, 2017-02-16:
Hmm spooky. I have a loop that runs an HTTP request to the IoT device, gets the list of RSSI's, sends it to the server and repeats. Since each request to the IoT device takes about 5-7 seconds (it scans and then responds...not ideal but it works) it is inserting the data in intervals of ~8 seconds. The server responds with a success message every time.

I am suspecting that it overwrites them since the dashboard (and the newly added `/locations` API) both respond with only one fingerprint stored. Note that I am calling `/calculate` after every call to `/learn` (for debugging purposes since I do not have a button to add a location just yet). The object I am sending to the server simply does not contain a timestamp. I changed it so that it adds the `time` property and now they get stored properly, hence the assumption of the overwriting.

schollz, 2017-02-16:
Ah! The number on the dashboard is somewhat misleading as it shows only the fingerprints for learning, and since it learns using 4-fold cross-validation. That is, you will have about four times the fingerprints as shown there, but only the ones shown on the dashboard are used for actual learning. Sorry that this is confusing, there is no documentation of this so there is no reason you should know about it...

TheMegaTB, 2017-02-16:
Ah okay. That makes sense. I was already wondering, it sometimes seemed to skip some fingerprints when learning using the app but that explains it. Thanks for the explanation there.

Just out of curiosity, is there like a 'recommended' minimum of training batches?
Would be nice to show the user some kind of hint telling him that for persistent results he should be inserting at least `n` fingerprints (I'm developing an UI for the ESP8266 currently).

closed 2017-02-16 by TheMegaTB