schollz/find #153: User does not show up at dashboard and /location API when omitting user=

TheMegaTB, 2017-02-22:
So I track by sending the following POST request:
```
POST /track HTTP/1.1
Host: 10.0.0.38:18003
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Content-Type: application/json
cache-control: no-cache
Content-Length: 524

{"group":"Family","username":"Til","wifi-fingerprint":[{"mac":"24:65:11:8A:B5:2B","rssi":-77},{"mac":"2E:D0:5A:C9:1F:AF","rssi":-77},{"mac":"5C:49:79:C3:7B:AB","rssi":-75},{"mac":"C8:0E:14:D9:39:A7","rssi":-92},{"mac":"88:03:55:E5:CD:4E","rssi":-93},{"mac":"E0:28:6D:0A:9F:8F","rssi":-93},{"mac":"06:24:FE:BC:10:9E","rssi":-92},{"mac":"34:81:C4:F9:0E:2A","rssi":-92},{"mac":"A6:81:C4:F9:0E:2A","rssi":-94},{"mac":"B6:81:C4:F9:0E:2A","rssi":-90},{"mac":"88:03:55:19:42:98","rssi":-96},{"mac":"78:8D:F7:BC:B3:2B","rssi":-92}]}
```

Now after doing that a few times I run a GET request to the /location API using a wildcard for the users (API doc states that user(s)= can be omitted to get everybody):
```
$ curl http://localhost:18003/location\?group\=family      
{"message":"No users found for username ","success":true,"users":null}
```

Running the same request with a user specified works:
```
$ curl http://localhost:18003/location\?group\=family\&user\=Til
{"message":"Correctly found locations.","success":true,"users":{"Til":[{"time":"2017-02-22 17:29:12.247730839 +0000 UTC","location":"bedroom","bayes":{"bedroom":1.3469514379659981,"diningroom":-0.6480348776801254,"kitchen":0.9413947866457815,"livingroom":-0.9434112338036101,"stairwell":-0.5519480265367511,"washroom":-0.1449520865912915},"svm":{"bedroom":-0.039694505375092025,"diningroom":-4.234959268096719,"kitchen":-5.249675784845004,"livingroom":-4.179967812196895,"stairwell":-6.076482276370932,"washroom":-6.442613449153309},"rf":{"bedroom":0.9,"diningroom":0.1,"kitchen":0,"livingroom":0,"stairwell":0,"washroom":0}}]}
```

schollz, 2017-02-25:
Hmmm I'm not sure I can replicate this error. I thought maybe its a cache issue, if you send a /track with a new user it still updates the cache. 

Can you try calling /calculate to see if it updates?

If the problem persists, send me your database file at hypercube.platforms@gmail.com

TheMegaTB, 2017-02-26:
Calling /calculate didn't fix it. One thing I noticed though is that in the data directory there are two files. One being uppercase `Family.db` and one lowercase `family.db`. Since the tracking request sends the group as uppercase and learning happened with it being lowercase could that possibly confuse something? The database files should be at your inbox.

schollz, 2017-02-27:
Yeah, you have to keep the case the same for the same group.

TheMegaTB, 2017-02-27:
Okay. I ignored that since it converted that to lowercase when learning but thanks for the information.

closed 2017-02-27 by TheMegaTB