schollz/find #154: OpenWRT

Pimmetje, 2017-02-25:
Hi i was wondering without reading the docs in too much detail (i am sorry if i overlooked it). But could openwrt be used with find like raspberry pi? This would allow for a very broad range of possible wifi data gather points. What data is needed from the wifi devices?

TheMegaTB, 2017-02-26:
The device basically needs to do a scan for other WiFi networks and send the resulting MAC -> RSSI values to a server. I'm not too familiar with OpenWRT but it should be possible to integrate it. I am not sure if that's what you are trying to achieve though since that would determine the location of the OpenWRT device, not of its clients.

Pimmetje, 2017-02-26:
No i would like to have the location of the wifi clients surrounding a few openwrt boxes like with the Raspberry Pi's. But what data does find need? Only single strength and MAC from all wifi devices or something special?

TheMegaTB, 2017-02-26:
Ah nvmd so you are referring to [this](https://github.com/schollz/find-lf) project instead. Afaik it is required to get the probe requests every WiFi devices sends out. Can't tell you any details though. You might want to have a look at the linked repo.

Pimmetje, 2017-02-26:
I missed that thanks

closed 2017-02-26 by Pimmetje