schollz/find #155: SVM model recalculation hangs indefinetely

shailesh17mar, 2017-03-03:
Since I dont understand the code completely, I will tell where everything hangs.
In your posteriorThreads.go file, you are looping over `numJobs' variable which for my case is 21879. Now this loop hangs on index 21807 and that's where everything stops working.
If I hardcode loop to 21807numJobs everything works flawlessly.
From my understanding this is because of the large size of finalResults hashmap.(I may be wrong completely).
Help me out here.
Also, Can you tell what exactly is happening in `function optimizePriorsThreaded`?

schollz, 2017-03-03:
Does it not hang if you turn off SVM?

What error are you seeing?

What system are you using?

shailesh17mar, 2017-03-04:
There's no error. It just hangs.
Haven't tried by turnning SVM off let me check that as well. 
Ubuntu 14.04 64bit / 8GB ram/ intel i3